AravisProject/aravis #109: ubuntu disconnection while working for a while normally

kmchinadragon, 2017-10-06:
I tried aravis both on ubuntu 16.4 on x86 and ubuntu14.04 on Jetson TK1(Nvidia). All seem to work correctly , such as, list the devices, get the information , and start to capture the pic. But , for about one or more minutes(seemed to be random), prompted that can not send cmd to the camera and quit. Howevr, right at this moment, IP address seemed to disappear and Ubuntu displayed the message that said "disconnected from xxxx".
     But when I tried it exactly the same on  AM5728 with the Embedded linux(not Ubuntu), it could work a very long time and no problem found.
      As derived from above, it seeded to be aravis something wrong with Ubuntu. For if I used the same ethernet port over the internet , it could work ok also.

EmmanuelP, 2017-10-25:
I don't think aravis does something weird with respect to the network stack. The information you gave don't allow me to understand what could go wrong with your setup.

Please attach any system log or aravis debug output that could help to narrow the issue.

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP