AravisProject/aravis #111: Constant timeouts after updating Aravis to 0.5.10 release and onwards

AntonioMichaelAlvarez, 2017-10-17:
Hello.     
I'm using Aravis library to capture frames from some Chinese camera DO3THINK GMT-200. It's a color camera, capable of streaming 1600x1200 @ 50 FPS video.    
Previously I was using aravis 0.5.9, but after moving to new PC, I've reinstalled OS (Xubuntu Linux 16.04.3 LTS) and downloaded current master branch of Aravis. After that I've encountered numerous buffer timeouts. I'm providing code for my program, maybe there's something wrong with program logic. This software is proven to work using Aravis 0.5.9 and OpenCV 3.2.0 (no extra modules required). Building instructions:

```sh
g++ -o switch_res switch_res.cpp -std=c++11 `pkg-config --cflags --libs opencv aravis-0.6`
```

```c++
#include "aravis-0.6/arv.h"
#include <opencv2/opencv.hpp>
#include <cstdlib>
#include <iostream>
#include <vector>
#include "signal.h"

double gain = 4;
unsigned int exposure_time = 10000;
bool pressed_flag = false;
bool is_running = true;
/*
 * Camera acquisition mode. Affects image resolution and number of transfered data.
 * Check values in camera's Genicam XML (Enumeration "ResolutionSel").
 * Default value is 0 (1600x1200).
 * R1600_1200_Normal 0
 * R800_600_bin      1
 * R1280_720_Normal  2
 * R800_600_Normal   3
 * R640_480_Normal   4
 * R320_240_Normal   5
 * custom            6
 */
int camera_mode = 0;
std::vector<std::pair<int, int>> cam_modes = {
    {1600, 1200},
    {800, 600},
    {1280, 720},
    {800, 600},
    {640, 480},
    {320, 240}
};

static void n_sig_handler(int i) {
    std::cout << "Snap! Caught SIGINT, exiting" << std::endl;
    is_running = false;
}

void camera_connect(const char *cam_address, ArvDevice **camera, ArvStream **stream, guint *payload) {
    arv_update_device_list();
    *camera = arv_gv_device_new(
            g_inet_address_new_from_string("192.168.1.77"), // PC's IP address
            g_inet_address_new_from_string(cam_address));
    if (*camera != NULL) {
        arv_device_set_integer_feature_value (*camera, "PixelFormat", ARV_PIXEL_FORMAT_MONO_8);
        arv_device_set_integer_feature_value (*camera, "ResolutionSelReg", camera_mode);
        arv_device_set_string_feature_value  (*camera, "ExposureAuto", arv_auto_to_string (ARV_AUTO_OFF));
        arv_device_set_string_feature_value  (*camera, "GainAuto", arv_auto_to_string (ARV_AUTO_OFF));
        arv_device_set_float_feature_value   (*camera, "ExposureTime", exposure_time);
        arv_device_set_float_feature_value   (*camera, "Gain", gain);
        
        *payload = arv_device_get_integer_feature_value (*camera, "PayloadSize");
        
        *stream = arv_device_create_stream (*camera, NULL, NULL);
        if(*stream != NULL) {
            arv_stream_push_buffer(*stream, arv_buffer_new(*payload, NULL));
            arv_device_execute_command (*camera, "AcquisitionStart");
        } else {
            std::cerr << "Error opening stream" << std::endl;
        }
    } else {
        std::cerr << "No camera available. Check connection to camera" << std::endl;
    }
}

void camera_disconnect(ArvDevice **camera, ArvStream **stream) {
    std::cout << "Disconnecting..." << std::endl;
    arv_device_execute_command (*camera, "AcquisitionStop");
    g_object_unref(*stream);
    g_object_unref(*camera);
    *stream = NULL;
    *camera = NULL;
}

int main(int argc, char** argv) {
    void (*old_sigint_handler)(int);
    old_sigint_handler = signal(SIGINT, n_sig_handler);
    ArvDevice *camera = NULL;
    ArvStream *stream = NULL;
    guint payload = 0;
    ArvBuffer *buffer;
    char* buffer_data;
    size_t buffer_size;
    int unsuccesfull_buffers = 0;
    int succesfull_buffers = 0;
    char k = 0;
    
    arv_g_type_init();
    
    cv::namedWindow("camera-0", CV_WINDOW_NORMAL);
    cv::Mat img;
    while (is_running) {
        if ((camera == NULL) || (stream == NULL)) {
            std::cout << "Connecting to 192.168.1.12 ..." << std::endl;
            camera_connect("192.168.1.12", &camera, &stream, &payload);
//            sleep(1);
        }
        else {
            // Main cycle
            buffer = arv_stream_timeout_pop_buffer(stream, 100000);
            if (buffer != NULL) {
                if (arv_buffer_get_status(buffer) == ARV_BUFFER_STATUS_SUCCESS) {
                    buffer_data = (char*)arv_buffer_get_data(buffer, &buffer_size);
                    if (buffer_size != payload) {
                        unsuccesfull_buffers++;
                    } else {
                        img = cv::Mat(cam_modes.at(camera_mode).second, cam_modes.at(camera_mode).first, CV_8UC1, buffer_data);
                        succesfull_buffers++;
                        unsuccesfull_buffers = 0;
                        cv::imshow("camera-0", img);
                        k = cv::waitKey(10);
                    }
                }
                if (is_running) {
                    arv_stream_push_buffer(stream, buffer);
                }
            } else {
                unsuccesfull_buffers++;
                if (unsuccesfull_buffers >= 10) {
                    std::cout << "10 unsuccessfull reads occured, reconnecting..." << std::endl;
                    camera_disconnect(&camera, &stream);
                }
            }
            switch(k) {
                case 'q':
                    is_running = false;
                    break;
                case 'w':
                    if (!pressed_flag) {
                        exposure_time += 1000;
                        if(exposure_time > 30000) exposure_time = 30000;
                        arv_device_set_float_feature_value(camera, "ExposureTime", exposure_time);
                        pressed_flag = true;
                    }
                    break;
                case 's':
                    if (!pressed_flag) {
                        exposure_time -= 1000;
                        if(exposure_time < 10000) exposure_time = 10000;
                        arv_device_set_float_feature_value(camera, "ExposureTime", exposure_time);
                        pressed_flag = true;
                    }
                    break;
                case 'e':
                    if (!pressed_flag) {
                        gain += 0.5;
                        if (gain > 8) gain = 8;
                        arv_device_set_float_feature_value(camera, "Gain", gain);
                        pressed_flag = true;
                    }
                    break;
                case 'd':
                    if (!pressed_flag) {
                        gain -= 0.5;
                        if (gain < 1) gain = 1;
                        arv_device_set_float_feature_value(camera, "Gain", gain);
                        pressed_flag = true;
                    }
                    break;
            case 'm':
                if (!pressed_flag) {
                    std::cout << "Current mode: " << camera_mode << std::endl;
                    camera_mode++;
                    if(camera_mode > 5) camera_mode = 0;
                    std::cout << "New mode: " << camera_mode << std::endl;

                    arv_device_set_integer_feature_value(camera, "ResolutionSelReg", camera_mode);
                    std::cout << "Old payload is " << payload << std::endl;
                    payload = arv_device_get_integer_feature_value (camera, "PayloadSize");
                    while((buffer = arv_stream_timeout_pop_buffer(stream, 100000)) != NULL);
                    arv_stream_push_buffer(stream, arv_buffer_new(payload, NULL));
                    std::cout << "New payload is " << payload << std::endl;

                    pressed_flag = true;
                }
                break;
                default:
                    pressed_flag = false;
                    break;
            }
        }
    }
    camera_disconnect(&camera, &stream);
    
    std::cout << "Exit" << std::endl;
    return (EXIT_SUCCESS);
}

```

AntonioMichaelAlvarez, 2017-10-17:
Attaching DO3THINK-DSGG200000025.xml containig Genicam data received from camera.
[DO3THINK-DSGG200000025.txt](https://github.com/AravisProject/aravis/files/1391556/DO3THINK-DSGG200000025.txt)

EmmanuelP, 2017-10-25:
In aravis 0.5.10, I have introduced a new `packet-request-ratio` parameter to arvgvstream:
https://github.com/AravisProject/aravis/commit/45928f7ee1de46d4a7bb7ba93f9980dd2012aef3

May be the default value is too low in your case.

Please reopen this report if increasing the value does not fix the your issue.

closed 2017-10-25 by EmmanuelP

AntonioMichaelAlvarez, 2018-01-18:
After two months, I've finally managed to test your suggestion.    
Indeed, increasing value `ARV_GV_STREAM_PACKET_REQUEST_RATIO_DEFAULT` (file `arvgvsttream.c`, line 63), makes test program work correctly, though I have an issue with modifying this value. Is there any method that I can call in my program to set ArvGvStream's property `ARV_GV_STREAM_PROPERTY_PACKET_REQUEST_RATIO` to desired value without modifying library's sources?

AntonioMichaelAlvarez, 2018-01-18:
This worked:
```c++
GValue packet_request_ratio = {0, };
g_value_init(&packet_request_ratio, G_TYPE_DOUBLE);
g_value_set_double(&packet_request_ratio, (gdouble) 0.5);
g_object_set_property(G_OBJECT(camera_stream), "packet-request-ratio", &packet_request_ratio);
```

AntonioMichaelAlvarez, 2018-01-18:
Thank you for your advice.