AravisProject/aravis #112: Modifying MakeFile to reference libpng

teegeecid, 2017-10-21:
Running example code such as arvcameratest runs fine, but I want to modify the code to save images. Including anything from libpng library, I get undefined references; not sure how to change the MAKEFILE to get it to be properly referenced.  I tried including -lpng to ARAVIS_LIBS, but that doesn't seem to help.

EmmanuelP, 2017-10-25:
A quick fix would be to add libpng in the aravis dependencies, here:

https://github.com/AravisProject/aravis/blob/a38cad9de14a00ebdf8e712b572b37825d408655/configure.ac#L95

ARAVIS_REQUIREMENTS="glib-2.0 >= 2.26 gobject-2.0 gio-2.0 libxml-2.0 gthread-2.0 libpng >= 1.6"

teegeecid, 2017-10-25:
thansk for replying!  I tried making the change you suggested, but I still get undefined references. 

Here's some additional details:

-Testing with Aravis 5.10, Ubuntu 16.04
-modified arvcameratest to include the code found here: https://github.com/szmoore/aravis-save-png

Is there an alternate method available that you would suggest in order to be able to save images with Aravis?

EmmanuelP, 2017-10-27:
> thansk for replying! I tried making the change you suggested, but I still get undefined references.

Could you attach the compilation output ?

EmmanuelP, 2017-10-27:
>  Is there an alternate method available that you would suggest in order to be able to save images with Aravis?

You may want to try to build a gstreamer pipeline.

teegeecid, 2017-10-27:
Hi Emmanuel,

This is what I get when I run sudo make on the make file provided with aravis :

Making all in src
make[1]: Entering directory '/home/user1/Desktop/aravis-0.5.10/src'
make  all-am
make[2]: Entering directory '/home/user1/Desktop/aravis-0.5.10/src'
make[2]: Nothing to be done for 'all-am'.
make[2]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/src'
make[1]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/src'
Making all in viewer
make[1]: Entering directory '/home/user1/Desktop/aravis-0.5.10/viewer'
Making all in data
make[2]: Entering directory '/home/user1/Desktop/aravis-0.5.10/viewer/data'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/viewer/data'
make[2]: Entering directory '/home/user1/Desktop/aravis-0.5.10/viewer'
make[2]: Nothing to be done for 'all-am'.
make[2]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/viewer'
make[1]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/viewer'
Making all in gst
make[1]: Entering directory '/home/user1/Desktop/aravis-0.5.10/gst'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/gst'
Making all in tests
make[1]: Entering directory '/home/user1/Desktop/aravis-0.5.10/tests'
  CCLD     arv-camera-test
arvcameratest.o: In function `arv_save_png':
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:246: undefined reference to `png_create_write_struct'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:247: undefined reference to `png_create_info_struct'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:249: undefined reference to `png_init_io'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:250: undefined reference to `png_set_IHDR'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:252: undefined reference to `png_write_info'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:255: undefined reference to `png_malloc'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:260: undefined reference to `png_write_image'
/home/user1/Desktop/aravis-0.5.10/tests/arvcameratest.c:261: undefined reference to `png_write_end'
collect2: error: ld returned 1 exit status
Makefile:851: recipe for target 'arv-camera-test' failed
make[1]: *** [arv-camera-test] Error 1
make[1]: Leaving directory '/home/user1/Desktop/aravis-0.5.10/tests'
Makefile:581: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

EmmanuelP, 2017-10-29:
It is the whole configuration/compilation log that I am interrested in.

After you have modified configure.ac, you should run ./autogen.sh, then ./configure, then make.

At the end of the configuration step, you should see the compilation flags used for the compilation of aravis. libpng should be present in the list.

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP