AravisProject/aravis #113: Long response times for SVS Vistek cameras

nerian-vision, 2017-10-24:
I have a set of SVS Vistek USB3Vision cameras (model: exo249Mu3), which report a very high maximum device response time of 5000 ms. This has the effect that when unreferencing the stream object, there will be a very long delay as the usb thread is blocking until the timeout.

I would propose to set a maximum allowed time out for the device response time. Could this break anything?

EmmanuelP, 2017-10-25:
I guess arv_uv_device_bulk_transfer could take another parameter for a different timeout than the one found in ARV_UVCP_DEFAULT_RESPONSE_TIME_MS.
That would allow to force it to a reasonable value in arvuvstream.c

nerian-vision, 2017-10-30:
I have solved this for me by adding an upper limit for the timeouts of bulk transfers:
https://github.com/nerian-vision/aravis/commit/6709f22b3d43213e147ad548de2eeafc45097a11

Could this be merged in the official version?

EmmanuelP, 2017-11-22:
I have commented the changes. Please reopen this issue or open a pull request if you have a new version.

closed 2017-11-22 by EmmanuelP