AravisProject/aravis #115: Cannot open video stream in viewer

cjtFD, 2017-10-27:
Hi,

I'm trying to use the arv-viewer to test two USB3-Vision cameras I've got here.
1. XIMEA MQ042CG-CM (https://www.ximea.com/products/usb3-vision-cameras-xiq-line/mq042cg-cm)
2. IDS UI-3580ML (https://en.ids-imaging.com/store/ui-3580ml.html)

There is only one of those cameras connected at a time. Both cameras seem to be detected by the viewer. But selecting one and pressing the play button results in a crash stating invalid memory access. Sometimes, if there isn't an instant crash, the header of the stream gui is shown but no image beneath it, only the taskbar with some buttons on it. Strangly the same happens to the "Fake_1" camera.

The later goal is to use the aravis module for OpenCV to receive an image to work with. But I always receive another error so I first wanted to test that the cameras work with the viewer.

EmmanuelP, 2017-10-27:
Could you attach a stack trace collected after the crash ?

```
gdb --args /path/to/arv-viewer
run
```

Then, after the crash:

```
thread apply all bt full
```

What platform are you using (OS, hardware) ?

cjtFD, 2017-10-27:
I've attached the log.
[logging.log](https://github.com/AravisProject/aravis/files/1422339/logging.log)

Currently running:
CentOS Linux release 7.3.1611 (Core)
Intel(R) Core(TM) i5-6442EQ CPU @ 1.90GHz
8 GB RAM

EmmanuelP, 2017-10-27:
It seems to have crashed in the bayer element. I guess it was obtained with a real camera. Could you attach a backtrace after a crash using the fake camera ?

cjtFD, 2017-10-27:
It indeed was a real camera. Here's the log from the fake one:
[fake-logging.log](https://github.com/AravisProject/aravis/files/1422523/fake-logging.log)

cjtFD, 2017-10-30:
Anything useful in this log?

EmmanuelP, 2017-10-30:
> Anything useful in this log?

Yes. Both crashes happened in different GStreamer functions that have the orc library in common.: [Orc - Optimized Inner Loop Runtime Compiler ](https://github.com/GStreamer/orc)

Did you compile GStreamer by yourself ? You may want to test your setup with a simple pipeline like this one:

` gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink`

cjtFD, 2017-10-30:
I've not compiled GStreamer myself. The command you linked worked fine as far as I can tell. A small windows poped up with some kind of test image and no errors were shown.

cjtFD, 2017-11-03:
I don't want to be annoying but I'd really like this to work. Any ideas?

EmmanuelP, 2017-11-03:
Oh, you are not annoying. It is just I have no idea what goes wrong on your system.

You may try to launch a pipeline using aravissrc, in the gst directory of aravis:

./gst/gst-aravis-launch aravissrc ! videoconvert ! autovideosink

cjtFD, 2017-11-03:
It returns an error:
WARNING: erroneous pipeline: no element "aravissrc"

EmmanuelP, 2017-11-03:
Oh sorry, it only works from the gst directory:

```
cd gst
./gst-aravis-launch aravissrc ! videoconvert ! autovideosink
```

EmmanuelP, 2017-11-03:
That means your camera was not found. Make sure it is up and available using:

```
arv-tool-0.6
```

cjtFD, 2017-11-03:
Oh, wasn't aware it needed to be connected for this. Here's the log after I did connect it.

```Leitung wird auf PAUSIERT gesetzt ... (setting pipeline to paused)
Leitung ist aktiv und erfordert keinen VORLAUF … (pipeline is live and needs no preroll)
FEHLER: Von Element /GstPipeline:pipeline0/GstAravis:aravis0: Interner Fehler im Datenfluss. (ERROR: In Element ... : Internal data flow error.)
Zusätzliche Fehlerdiagnoseinformation: (Additional debug info:)
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming task paused, reason not-negotiated (-4)
FEHLER: Leitung läuft nicht vor. (Not sure what the correct translation is, maybe something like ERROR: Pipeline isn't prerolling.)
Leitung wird auf PAUSIERT gesetzt ... (setting pipeline to paused)
Leitung wird auf BEREIT gesetzt ... (setting pipeline to ready)
Leitung wird auf NULL gesetzt ... (setting pipeline to null)
Leitung wird geleert ... (freeing pipeline)
```

Btw: output of the arv-tool was: 
`XIMEA-45683851 (USB3)`

EmmanuelP, 2017-11-03:
About the german logs, it is better to set LANG to C:
```
export LANG=C
```

I don't know why gstreamer was not able to start the pipeline. May be a pixel format issue. What gives:
```
arv-tool-0.6 features
```

cjtFD, 2017-11-03:
Ok, thanks.

[arv-tool-0.6 features](https://github.com/AravisProject/aravis/files/1441437/features.txt)

EmmanuelP, 2017-11-03:
Supported format are all bayer, which requires an additional element in the gstreamer pipeline. This one should work:
```
./gst-aravis-launch aravissrc ! bayer2rgb ! videoconvert ! autovideosink
```

cjtFD, 2017-11-03:
This opened a window but nothing showed up. It basically took a screenshot of my terminal instead of showing the cameras image. It actually feels like using the arv-viewer, were basically only the taskbar is visible.
[Output](https://github.com/AravisProject/aravis/files/1441513/output.txt)

EmmanuelP, 2017-11-04:
Could you try:
```
gst-launch-1.0 videotestsrc ! videoconvert ! rgb2bayer ! bayer2rgb ! videoconvert ! autovideosink
```
Also, could you run arv-camera-test:
```
./tests/arv-camera-test --debug=all:3
 ```

cjtFD, 2017-11-06:
Output looked the same as the previous gst launch only the window was smaller. [gst launch.txt](https://github.com/AravisProject/aravis/files/1445337/gst.launch.txt)

How long should i leave the tests running for? The debug file size is already 250mb.
Btw: The LED on the camera that is supposed to be showing if a picture is taken is flashing during the tests if that's any useful.

EmmanuelP, 2017-11-06:
> Output looked the same as the previous gst launch only the window was smaller. gst launch.txt

Black window ?

> How long should i leave the tests running for? The debug file size is already 250mb and is running for more than a hour.

Just run arv-camera-test 10s, it is enough.

cjtFD, 2017-11-06:
> Black window ?

No, it basically takes a screenshot of whatever is behind the opening window and displays that the whole time. If I try to take a screenshot to show you the windows output it's just black, so I can't show you what I mean specifically.

[testoutput](https://github.com/AravisProject/aravis/files/1445749/testoutput.txt)

EmmanuelP, 2018-08-10:
Everything suggests it is an issue in the gstreamer bayer code.

I'm closing this issue. Please reopen if you have additional informations.

closed 2018-08-10 by EmmanuelP