AravisProject/aravis #116: FAILED Invalid Genicam XML data on U3 device

hdevillepoix, 2017-10-29:
Working with a USB3 camera on Debian 9, using the Aravis master branch.
```
$ arv-test-0.6 
Testing device 'Blackfly BFLY-U3-23S6M' from 'Point Grey Research'
(process:4963): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
(process:4963): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
Genicam: FAILED Invalid Genicam XML data
Sensor size read: 1920 1200
```
Any idea of where the problem might lie?

I somewhat manage to capture with the camera, but I get problems like `(process:5126): Aravis-CRITICAL **: arv_device_get_float_feature_bounds: assertion 'ARV_IS_DEVICE (device)' failed` when trying to call `arv_camera_get_exposure_time_bounds`.

EmmanuelP, 2017-10-30:
arv-tool is an embryonic project of an automated testing software. The critical messages are expected. The fact arv-tool returns a correct sensor size means the genicam data are correctly loaded.

> I somewhat manage to capture with the camera

What does that mean ? Is it an attempt to use the aravis API ? Did you try to enable aravis debug output by setting  ARV_DEBUG enivronment variable to `all:3` ?

Did you succeed in visualizing the video stream using arv-viewer ?

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP