AravisProject/aravis #118: ARV_BUFFER_STATUS_TIMEOUT and Frame Rate 0

haseeb-uob, 2017-11-10:
I am getting error ARV_BUFFER_STATUS_TIMEOUT . I also tried to increase timeout but still the same error.
Please have a look at output of ./tests/arv-camera-test ./arv-camera-test -d stream:3.

[arv_camera_test.txt](https://github.com/AravisProject/aravis/files/1461122/arv_camera_test.txt)

EmmanuelP, 2017-11-14:
The log output says aravis is only receiving a small part of the frame packets.

- check the network bandwidth allows to transmit the video stream
- try the automatic buffer size option (--auto)
- try to enable packet socket support
-  try --realtime or --high-priority

EmmanuelP, 2017-11-22:
Did you succeed in getting completed frames ?

waspinator, 2017-12-13:
I have a similar issue when trying to use the ROS (robot operating system) node. Running `/tests/arv-camera-test` seems to work, I don't see any failures or underruns, so I'm guessing that the ROS implementation has an issue. 

I'm not very familiar with aravis, so if there is a glaring issue in the code, I would really appreciate it if you would point it out.

Thanks

https://github.com/waspinator/camera_aravis/blob/ffd161e9cb7b3bf9f9d32e3f6b086f66d7b9325d/src/camnode.cpp#L157

EmmanuelP, 2018-08-10:
@waspinator I have not detected something wrong in your code (quick glance). Did you try to implement one or several suggestions of comment https://github.com/AravisProject/aravis/issues/118#issuecomment-344266129 ?

EmmanuelP, 2018-08-10:
@haseeb-uob No answer. I'm closing this issue. Please reopen if you have the requested informations.

Cheers.

closed 2018-08-10 by EmmanuelP