AravisProject/aravis #124: Impossible to exclude building of documentation

Bidski, 2017-12-28:
I am trying to cross compile aravis for a different system and I would also like the build to be as minimal as possible (only the essential core parts of the library, no viewer, no documentation, no tests).

Unfortunately, it seems that autogen.sh assumes that I want to build documentation and that I have the tools built to do this (I do not, and I do not want to build them either).

It also appears that the 0_5_10 release has removed the configure options to disable building of documentation
```
configure: WARNING: unrecognized options: --disable-gtk-doc, --disable-gtk-doc-html, --disable-gtk-doc-pdf
```

Is it possible to fix this so that a minimal build can be achieved?

Bidski, 2017-12-28:
Following up on this, is it also possible for the releases to have a generated configure file in them? This seems to be a common thing in a lot of other libraries that are using the autotools build system, but not this library.

EmmanuelP, 2017-12-28:
Blame github that automatically creates 'Releases' from git tags. I don't know how to disable this behaviour. It looks like I'm able to delete these fake releases though, so that is what I will do.

Real release tarballs, with documentation already built, can be found here:

http://ftp.gnome.org/pub/GNOME/sources/aravis/0.5/

They include a configure script.

Pleasse reopen this issue if you still have issues releated to documentation building.

closed 2017-12-28 by EmmanuelP