AravisProject/aravis #125: arv-viewer Improve handling of buffer objects

TIS-Tim, 2018-01-08:
While testing arv-viewer with high-resolution U3V cameras on a somewhat weak system, I came across an issue in the way arv-viewer handles the arv_buffer and GstBuffer objects.

Testing showed that the gstreamer pipeline can hold more than one buffer, but arv-viewer currently assumes that the pipeline contains exactly one, and returns the buffer to aravis before gstreamer is done with it, leading to corruption of the displayed image with new data.

Here is a diff of my proposed changes:
[arv-viewer.diff.zip](https://github.com/AravisProject/aravis/files/1610983/arv-viewer.diff.zip)

It uses the notification callback from GstBuffer to return the buffer to aravis, which should be the right point in time.

(I am unsure about last_buffer still being used for the snapshot function, since it might still suffer from the same corruption problem)

EmmanuelP, 2018-01-09:
Hi.

Thanks for the report.

It would be nice if you could create a pull request when you have a patch proposal. It makes review easier.

I know this was an issue for a while, but never took time to fix it properly, so thanks for your proposal. It mostly looks fine, but I'm worried about the lifetime of the ArvStream pointed by cb_param->stream. It could be better to keep a weak reference on it to be sure it is still valid in the callback context.

TIS-Tim, 2018-01-09:
Will do

EmmanuelP, 2018-01-11:
Here is the corresponding pull request:

https://github.com/AravisProject/aravis/pull/128

closed 2018-01-11 by EmmanuelP