AravisProject/aravis #127: GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

AjayGarad, 2018-01-08:
Hi,
I have contrastech mars 640-815UC camera(USB 3.0).
When I "run - arv-tool-0.6"
I got the following answer:
ContrasTech-3D02158AAK00012 (USB3)

then I run "arv-tool-0.6 features" and I got the following error:

ContrasTech-3D02158AAK00012 (USB3)
(arv-tool-0.6:16863): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
(arv-tool-0.6:16863): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Device 'ContrasTech-3D02158AAK00012' not found

I tried to run the "./arv-camera-test" but still got the same error:

Looking for the first available camera
(lt-arv-camera-test:16846): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
No camera found

Can you tell why I am getting this error?

EmmanuelP, 2018-01-08:
Please attach the output of:

`arv-tool-0.6 features -d all:3`

AjayGarad, 2018-01-09:
This is the output I got:

Found 1 USB3Vision device (among 9 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.6.180
ContrasTech-3D02158AAK00012 (USB3)
[UvDevice::new] Vendor  = ContrasTech
[UvDevice::new] Product = Mars640-815uc
[UvDevice::new] S/N     = 3D02158AAK00012
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 3, interface 2
[UvDevice::new] Failed to claim USB interface to 'Mars640-815uc - #3D02158AAK00012'

(arv-tool-0.6:5837): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Found 1 USB3Vision device (among 9 USB devices)
[UvDevice::new] Vendor  = ContrasTech
[UvDevice::new] Product = Mars640-815uc
[UvDevice::new] S/N     = 3D02158AAK00012
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 3, interface 2
[UvDevice::new] Failed to claim USB interface to 'Mars640-815uc - #3D02158AAK00012'

(arv-tool-0.6:5837): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.6.180
Device 'ContrasTech-3D02158AAK00012' not found

EmmanuelP, 2018-01-09:
It looks like a permission issue.

Please try to run arv-test as root. If it works as root, then you can make the camera work as user by creating a udev rules file as described here: https://github.com/AravisProject/aravis#usb-permissions

AjayGarad, 2018-01-09:
I did lsusb and copied the ID in the udev rule file and copied the file in the location "/etc/udev/rules.d"
still getting error:
"sudo arv-test-0.6"
output: 
Testing device 'Mars640-815uc' from 'ContrasTech'

(process:10876): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:10876): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:10876): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:10876): Aravis-CRITICAL **: arv_device_get_genicam_xml: assertion 'ARV_IS_DEVICE (device)' failed

(process:10876): Aravis-CRITICAL **: arv_camera_get_sensor_size: assertion 'ARV_IS_CAMERA (camera)' failed

(process:10876): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:10876): Aravis-CRITICAL **: arv_device_get_status: assertion 'ARV_IS_DEVICE (device)' failed
Genicam: FAILED Failed to retrieve Genicam data
Sensor size read: FAILED

(process:10876): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

EmmanuelP, 2018-01-09:
Could you attach the output with -d all:3 please.

AjayGarad, 2018-01-09:
"sudo arv-test-0.6 -d all:3"

OUTPUT:
Testing device 'Mars640-815uc' from 'ContrasTech'

(process:2656): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:2656): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:2656): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:2656): Aravis-CRITICAL **: arv_device_get_genicam_xml: assertion 'ARV_IS_DEVICE (device)' failed

(process:2656): Aravis-CRITICAL **: arv_camera_get_sensor_size: assertion 'ARV_IS_CAMERA (camera)' failed

(process:2656): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:2656): Aravis-CRITICAL **: arv_device_get_status: assertion 'ARV_IS_DEVICE (device)' failed
Genicam: FAILED Failed to retrieve Genicam data
Sensor size read: FAILED

(process:2656): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

also,  sudo ./arv-camera-test -d all:3

OUTPUT:
Looking for the first available camera
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = ContrasTech
[UvDevice::new] Product = Mars640-815uc
[UvDevice::new] S/N     = 3D02158AAK00012
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 3, interface 2
[UvDevice::new] Failed to claim USB interface to 'Mars640-815uc - #3D02158AAK00012'

(lt-arv-camera-test:2989): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.12.146
[GvDiscoverSocket::new] Add interface 172.16.12.113
No camera found

"sudo arv-tool-0.6  -d all:3"

OUTPUT:
Found 1 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.12.146
[GvDiscoverSocket::new] Add interface 172.16.12.113
ContrasTech-3D02158AAK00012 (USB3)

EmmanuelP, 2018-01-09:
The output of `sudo arv-tool-0.6 -d all:3 features` I mean.

AjayGarad, 2018-01-09:
After running this:
~/software/aravis-0.5.10/tests$ sudo arv-tool-0.6  -d all:3 features

I got the following output:

Found 1 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.12.113
[GvDiscoverSocket::new] Add interface 172.16.12.146
ContrasTech-3D02158AAK00012 (USB3)
[UvDevice::new] Vendor  = ContrasTech
[UvDevice::new] Product = Mars640-815uc
[UvDevice::new] S/N     = 3D02158AAK00012
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 3, interface 2
[UvDevice::new] Failed to claim USB interface to 'Mars640-815uc - #3D02158AAK00012'

(arv-tool-0.6:11350): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = ContrasTech
[UvDevice::new] Product = Mars640-815uc
[UvDevice::new] S/N     = 3D02158AAK00012
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 3, interface 2
[UvDevice::new] Failed to claim USB interface to 'Mars640-815uc - #3D02158AAK00012'

(arv-tool-0.6:11350): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.16.12.113
[GvDiscoverSocket::new] Add interface 172.16.12.146
Device 'ContrasTech-3D02158AAK00012' not found

EmmanuelP, 2018-01-11:
I don't understand what goes wrong. It doesn't look like a permission issue. It is perhaps a configuration issue, when aravis tries to establish the connection to the device.

Did you try to connect to the camera using the manufacturer iCentral software ? (see issue #123)

AjayGarad, 2018-01-28:
Yes I tried with iCentral software, it was working.

madMAx43v3r, 2018-03-04:
I got similar errors when opening multiple USB3 cameras from different threads in parallel. The reason is the USB3 interface code is not thread-safe (all the functions that operate on the `ArvUvInterface` static singleton instance). To solve the issue I added a mutex in my code around calling `arv_camera_new(...)`.
So maybe this is also a multi-threading issue.

EmmanuelP, 2018-08-10:
@madMAx43v3r It would be nice if you could open a new issue explaining what you have seen, and which part of ArvInterface you think is not thread safe.

Thanks.

EmmanuelP, 2018-08-10:
@AjayGarad Did you finally succeed in connecting to your camera ?

madMAx43v3r, 2018-08-15:
@EmmanuelP Looks like you made it thread-safe on May 29 with commit 520bcc8560f412c01337835c7391f662f9bd6944

EmmanuelP, 2023-05-19:
Long time, no new. I'm closing this issue. Don't hesitate to reopen if you have the requested information.

closed 2023-05-19 by EmmanuelP