AravisProject/aravis #130: Debian package for aravis library

vincentrou, 2018-01-26:
Is there any project for packaging this library ?

EmmanuelP, 2018-01-26:
The subject was discussed, but there is nothing in debian or ubuntu yet:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=843185
https://launchpad.net/~vincentrou/+archive/ubuntu/ppa

I'm closing this issue as, even if I welcome warmly any effort to package aravis in distributions,  I don't think it is a good idea to maintain them here.

Cheers,

    Emmanuel.

closed 2018-01-26 by EmmanuelP

gavanderhoorn, 2019-12-19:
Just a comment for future readers: Ubuntu is shipping `v0.6.0` starting with Disco (ie: `19.04`): [libaravis-dev](https://packages.ubuntu.com/search?keywords=libaravis-dev).

Not sure whether this is an import from Debian or an Ubuntu packaging effort.

EmmanuelP, 2019-12-19:
> Not sure whether this is an import from Debian or an Ubuntu packaging effort.

It happened thanks to @cmarmo and the Debian Astronomy Team.

gavanderhoorn, 2019-12-19:
> > Not sure whether this is an import from Debian or an Ubuntu packaging effort.
> 
> It happened thanks to @cmarmo and the Debian Astronomy Team.

Ok. So that sounds like an import from upstream Debian then.

:+1: @cmarmo.