AravisProject/aravis #133: Use of buffer in Aravis

mehtajaghvi, 2018-02-02:
In the new issue, please add the following informations:

1. what you mean by bad frame (corrupted images ?): my images are not what I expect them to be. I trigger my camera based on an event. The initial few images are what I expect them to be but after few images it begins to show a completely different view  and the behavior is consistent. If I re run my python code, the images go back to normal (expected image) again for certain frames. This made me think it is the buffer. 
Solved: I set my stream to buffer size 1 during creation of camera object (I am not doing a video stream, just one frame at a time). I start acquisition, pop and push and stop camera acquisition in a while loop. But the issue persists. Finally, I re-create my stream for every image capture I do using aravis.create_stream(none,none) and that seems to solve the issue.

language used: **Python**
aravis version: **0.4**
Console output of your code with ARV_DEBUG environment variable set to stream:3
****export ARV_DEBUG=stream:3,device:3....I tried this but it didn't output anything****

You have to pop/push because aravis stores the frame it gets in a circular buffer. The use of a circular buffer helps to not constantly alloc/free memory: I understood this part, after working through the buffer   output.  But, is there a way to clear the buffer if I wanted to.

EmmanuelP, 2018-02-02:
I guess you may be are using and displaying the image buffer after the call to push_buffer. In this case, the buffer may be overwritten with a new frame before it is actually displayed on screen.

For my information, could you please attach the output of `arv-camera-test -d all,device:3` ? Just let it run for a few seconds.

mehtajaghvi, 2018-02-02:
self.camera.start_acquisition()	
       
	buffer = self.stream.pop_buffer()

	#print("Emite signals", self.stream.get_emit_signals())
	#self.capturetime = buffer.get_timestamp()
	#print("capture time", self.capturetime)
	
	if(buffer):
		data = buffer.get_data() 
		self.stream.push_buffer(buffer)

I am pushing after I get my data

EmmanuelP, 2018-02-03:
I don't know well how the python binding behaves, but if data is not a copy of the content pointed by buffer.get_data (), data may be overwritten by a new frame, because buffer is reused for stream reception.

If you want to be sure data is not overwritten, you should call push_buffer after you are done with data.

EmmanuelP, 2018-05-17:
May I close this report ?

closed 2018-05-24 by EmmanuelP