AravisProject/aravis #135: Cannot make a GigE camera work with a direct connection

RhiobeT, 2018-02-15:
Hi,

As a part of a project, I need to use a DMK 23GM021 camera with a Raspberry Pi.

After spending the whole day trying all the things I could think about, I managed to have the camera detected by the Aravis tools when plugged to my home router.
It also works well when plugged directly to a Windows computer.
However, I just can't get it to be detected when plugged directly to the Raspberry Pi, or to any other computer running Linux actually.

It may very well be a network issue, but I'm far from being competent in networking and I just can't tell what could me missing here.
I tried to setup a DHCP server on the Raspberry but it never receives any DHCPREQUEST messages from the camera (also, do I even need a DHCP server?)
The only packets I see with Wireshark are DISCOVERY_CMD, plus the DHCPDISCOVER and DHCPOFFER messages.

Some help would be great, thanks in advance~~

medasani456, 2018-02-19:
Do you install GigE drivers of the camera from TIS in the Raspberry Pi or Linux Machine ??

medasani456, 2018-02-19:
I have been working with Allied vision cameras in Ubuntu. Initially, I was unable to detect the camera in the arv-viewer after installing the gigE driver I am able to view the image from the camera.

It could be one reason.

RhiobeT, 2018-02-19:
Are you talking about the gige-daemon? Because it is indeed running on the Pi :/

medasani456, 2018-02-19:
I don't think this one. It will be something like ( VimbaGigETL for Allied Vision Cameras). 
Could be similar with an extension of TL(Transport Layer).

RhiobeT, 2018-02-19:
I don't think there is any other specific driver for TIS cameras...
I will try to post this issue on the tiscamera repository and get back if I find anything useful

RhiobeT, 2018-02-19:
Well, it turns out the culprit was the Pi's Ethernet port...
A GigE device needs, as its name implies, a speed of at least 1gb/s, and the port on the Pi is limited to 100mb/s, so it could hardly work :/

You can close the issue.

EmmanuelP, 2018-02-19:
Ok. Thanks.

closed 2018-02-19 by EmmanuelP