AravisProject/aravis #138: Under which license is Aravis published?

tglander, 2018-03-09:
The file COPYING contains the GPL license, i.e., a copyleft license which prohibits use of Aravis without also publishing derived code.
However, the README.md states that "Aravis is released under the LGPL v2+", which is a permissive license allowing derived to be closed source.
Which license is it?

EmmanuelP, 2018-03-09:
https://github.com/AravisProject/aravis/blob/d0c7ff2083d9abc22b70f4874cdad73c66cd99b8/COPYING#L1
A proprietary software can be linked to the aravis library.
Any distribution of a modified binary version of aravis must give access to the corresponding source code.

closed 2018-03-09 by EmmanuelP

tglander, 2018-03-09:
Thanks for clarifying. So it is GPL2, not LGPL.v2+. You might want to update the README.md to make it consistent.

EmmanuelP, 2018-03-09:
No. Gnu **L**ibrary **G**eneral **P**ublic **L**icense.

This is the v2 version. Headers say the code is licensed under the v2 or any later version. Hence v2+.