AravisProject/aravis #141: ArvStream thread clean up

rodolfo-picoreti, 2018-03-13:
Hi, I am trying to figure out how to properly clean up the thread created by `arv_camera_create_stream`. First I tried passing the ownership of the stream pointer to the callback function, but when I try to unref it inside the callback I get a deadlock error from glib. Then I tried unrefing it in the main thread after unrefing the buffer but still no success. I get warnings from glib saying that the unref failed. 

Some C++ code for context:

```cpp
struct Camera {
  ArvCamera* camera;
  ArvStream* stream;
  std::atomic<bool> shutdown;

  Camera() : shutdown(false) {
    camera = arv_fake_camera_new(nullptr);
    assert(camera);

    stream = arv_camera_create_stream(camera, on_buffer_event, this);
    assert(stream);

    arv_stream_push_buffer(stream, arv_buffer_new(arv_camera_get_payload(camera), nullptr));
    arv_camera_start_acquisition(camera);
  }

  ~Camera() {
    printf("destroying\n");
    arv_camera_stop_acquisition(camera);
    shutdown = true;
    // for simplicity just sleep why we wait for the buffer unref
    std::this_thread::sleep_for(std::chrono::seconds(2));

    g_object_unref(stream);
    g_object_unref(camera);
  }

  static void on_buffer_event(void* args, ArvStreamCallbackType type, ArvBuffer* buffer) {
    printf("called\n");
    auto self = static_cast<Camera*>(args);
    if (type == ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE) {
      if (self->shutdown) {
        g_object_unref(buffer);
        printf("buffer freed\n");
        return;
      }

      arv_stream_push_buffer(self->stream, buffer);
    }
  }
};
```

To test that some memory is being leaked I put the code on a loop constructing and destructing a Camera object and watch the memory growing. I tried using clangs memory sanitizer but it didn't catch anything even though I can see the memory growing.

```cpp
int main(int argc, char** argv) {
  for (;;) {
    Camera camera;
    std::this_thread::sleep_for(std::chrono::seconds(2));
  }
}
```

EmmanuelP, 2018-03-16:
In the on_buffer_event callback, you are pushing a buffer to the stream input queue or releasing it without removing it from the output queue before.

Also, you should not rely on a sleep function to wait for acquisition completion, but signal to the main thread, from the callback function, the acqusition is done.

closed 2018-03-16 by EmmanuelP

rodolfo-picoreti, 2018-03-16:
Hey Emmanuel thanks for the answer, 

So I'm assuming that by removing the buffer from the output queue you mean calling one of the stream_pop functions. I added `arv_stream_try_pop_buffer` like shown in the code below and it fixed the gobject unref assert errors. I've also added the basic synchronization code for completion. Even though I don't get any errors now the memory is still being leaked. Any idea of what I am doing wrong now?

Code:
```cpp
struct Camera {
  ArvCamera* camera;
  ArvStream* stream;

  enum State { kStreaming = 0, kWaitingShutdown = 1, kDone = 2 };

  State state;
  std::condition_variable cv;
  std::mutex mutex;

  Camera() : state(kStreaming) {
    camera = arv_camera_new(nullptr);
    assert(camera != nullptr);
    stream = arv_camera_create_stream(camera, on_buffer_event, this);
    assert(stream != nullptr);
    arv_stream_push_buffer(stream, arv_buffer_new(arv_camera_get_payload(camera), nullptr));
    arv_camera_start_acquisition(camera);
  }

  ~Camera() {
    std::unique_lock<std::mutex> guard(mutex);
    state = kWaitingShutdown;
    arv_camera_stop_acquisition(camera);
    cv.wait(guard, [&]() { return state == kDone; });

    g_object_unref(stream);
    g_object_unref(camera);
  }

  static void on_buffer_event(void* args, ArvStreamCallbackType type, ArvBuffer*) {
    auto self = static_cast<Camera*>(args);
    if (type == ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE) {
      auto buffer = arv_stream_try_pop_buffer(self->stream);
      assert(buffer != nullptr);

      std::unique_lock<std::mutex> guard(self->mutex);
      if (self->state == kWaitingShutdown) {
        g_object_unref(buffer);
        self->state = kDone;
        self->cv.notify_all();
        return;
      }

      arv_stream_push_buffer(self->stream, buffer);
    }
  }
};
```

Thanks again!

rodolfo-picoreti, 2018-03-16:
Trying to isolate the root cause I've made a minimal example and I no longer thing the thread cleanup is the problem here. Maybe I did'nt understand how to properly remove the buffer from the queue since the following code snippet is also reproducing the memory leak:

```cpp
void on_buffer_event(void* args, ArvStreamCallbackType type, ArvBuffer*) {
  auto stream = static_cast<ArvStream**>(args);
  auto buffer = arv_stream_try_pop_buffer(*stream);
  if (buffer != nullptr) { arv_stream_push_buffer(*stream, buffer); }
}

int main(int argc, char** argv) {
  ArvCamera* camera = arv_camera_new(nullptr);
  assert(camera != nullptr);

  ArvStream* stream;
  stream = arv_camera_create_stream(camera, on_buffer_event, &stream);
  assert(stream != nullptr);

  arv_stream_push_buffer(stream, arv_buffer_new(arv_camera_get_payload(camera), nullptr));
  arv_camera_start_acquisition(camera);

  for (;;) {
    std::this_thread::sleep_for(std::chrono::seconds(1));
  }
  // .... cleanup, etc
}
```