AravisProject/aravis #143: Cannot set camera region offsets

cmarmo, 2018-03-22:
Hi,
I'm working in implementing cropped buffer acquisition in an aravis based software and I'm having some troubles.
[This gist](https://gist.github.com/cmarmo/870638c88d1ef68831af9d91ce231e35) contains the code I'm using to check aravis behaviour.
My camera is a [DMK23G445](https://www.theimagingsource.com/products/industrial-cameras/gige-monochrome/dmk23g445/), and I'm linking to the aravis commit d0c7ff2083d9abc22b70f4874cdad73c66cd99b8
The image acquired with default region is 
[image]
The one acquired setting width = 640, height = 480 is
[image]
Apparently, dimensions are correctly set, but the the buffer is messed up in the x dimension: am I doing something wrong?
In addition to that, when I try to set x and y offset different from zero with `arv_camera_set_region()` then get the new values with `arv_camera_get_region()` I always obtain 0,0. Is that the expected behaviour?
Thanks for listening (and for your code!).
Best,
Chiara

cmarmo, 2018-03-27:
Hello Emmanuel,
and thanks for your answer.

Yes, I've tried with arv-viewer: `region size` and `region position` are just not taken into account... :(
I can't check on other cameras today, but indeed `arv-tool-0.6 genicam` on the DMK does not output  offset parameters.

Here are the `ImageFormat` parameters from DMK23G445
`  <Category Name="ImageFormat" NameSpace="Standard">
    <pFeature>SensorWidth</pFeature>
    <pFeature>SensorHeight</pFeature>
    <pFeature>Width</pFeature>
    <pFeature>Height</pFeature>
    <pFeature>PixelFormat</pFeature>
  </Category>`
So, maybe I have to check on offset availability in the genicam output?

EmmanuelP, 2018-03-27:
Please attach the output of `arv-tool-0.6 features`.

EmmanuelP, 2018-03-27:
So it seems it is not possible to set the offsets. That doesn't explain why the image is distorded though. May be there is som constraints to respect when setting the image size. Please refer to the camera documentation.

I'm afraid there is not a lot we can do in aravis. I'm closing this issue. Don't hesitate to reopen it if you think there is something to change in aravis.

Thanks.

Cheers.

closed 2018-03-27 by EmmanuelP