AravisProject/aravis #144: Problem with arv_camera_new()

ArkadiuszRaj, 2018-04-06:
Calling arv_camera_new() function fails internally.
Below is a extract of the ARV_DEBUG=all:3 log

```
[UvDevice::new] Product = Chameleon3 CM3-U3-31S4M
[UvDevice::new] S/N     = 010AFC78
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB interface to 'Chameleon3 CM3-U3-31S4M - #010AFC78'

(process:18078): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = Point Grey Research
[UvDevice::new] Product = Chameleon3 CM3-U3-31S4M
[UvDevice::new] S/N     = 010AFC78
20180406T183827UT : ACQ   0.0fps G:0.00 E:0us >>[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB interface to 'Chameleon3 CM3-U3-31S4M - #010AFC78'

(process:18078): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.1.200

```

ArkadiuszRaj, 2018-04-06:
Additional info:

`
# arv-test-0.6 
[XmlSchema::new_from_file] Invalid xsd file 'file:///media/git/PFNMCv2/build/aarch64/GenApiSchema_Version_1_0.xsd'
[XmlSchema::new_from_file] Invalid xsd file 'file:///media/git/PFNMCv2/build/aarch64/GenApiSchema_Version_1_1.xsd'
Found 1 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.1.200
Testing device 'Chameleon3 CM3-U3-31S4M' from 'Point Grey Research'
[UvDevice::new] Vendor  = Point Grey Research
[UvDevice::new] Product = Chameleon3 CM3-U3-31S4M
[UvDevice::new] S/N     = 010AFC78
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB interface to 'Chameleon3 CM3-U3-31S4M - #010AFC78'

(process:18146): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = Point Grey Research
[UvDevice::new] Product = Chameleon3 CM3-U3-31S4M
[UvDevice::new] S/N     = 010AFC78
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB interface to 'Chameleon3 CM3-U3-31S4M - #010AFC78'

(process:18146): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.1.200

(process:18146): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:18146): Aravis-CRITICAL **: arv_device_get_genicam_xml: assertion 'ARV_IS_DEVICE (device)' failed

(process:18146): Aravis-CRITICAL **: arv_camera_get_sensor_size: assertion 'ARV_IS_CAMERA (camera)' failed

(process:18146): Aravis-CRITICAL **: arv_camera_get_device: assertion 'ARV_IS_CAMERA (camera)' failed

(process:18146): Aravis-CRITICAL **: arv_device_get_status: assertion 'ARV_IS_DEVICE (device)' failed
Genicam: FAILED Failed to retrieve Genicam data
Sensor size read: FAILED

(process:18146): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

`

EmmanuelP, 2018-04-11:
> [UvDevice::new] Failed to claim USB interface to 'Chameleon3 CM3-U3-31S4M - #010AFC78'

Looks like a permission issue. Did you try using the root account ?

ArkadiuszRaj, 2018-04-19:
Yes with root works smoothly, but w/o sometimes ok, sometimes not.
Udev access is set.

EmmanuelP, 2018-08-10:
May be there is two different issues.

Regarding the permission one, did you create the relevant entry for your device in /etc/udev/rules.d ?

https://github.com/AravisProject/aravis#usb-permissions

If so, did it fix the permission issue ?

EmmanuelP, 2019-09-04:
So it looks like there is little we can do in Aravis.

I'm closing this issuse.Please reopen if you think there is something to fix in Ararvis.

Cheers,

closed 2019-09-04 by EmmanuelP