AravisProject/aravis #145: Camera only enumerates Continuous feature

mehtajaghvi, 2018-04-10:
Hi I am trying to put the camera into single-frame mode

Code ran
1.    features=self.get_feature_vals("AcquisitionMode")
        print("features",features)
       
    def get_feature_type(self, name):
        genicam = self.dev.get_genicam()
        node = genicam.get_node(name)
        if not node:
            raise AravisException("Feature {} does not seem to exist in camera".format(name))
        return node.get_node_name()
        
    def get_feature_vals(self, name):
        """
        if feature is an enumeration then return possible values
        """
        ntype = self.get_feature_type(name)      
        if ntype == "Enumeration":
            return self.dev.get_available_enumeration_feature_values_as_strings(name)
        else:
            raise AravisException("{} is not an enumeration but a {}".format(name, ntype))
Output:
The output f the above code is "Continuous" 
('features', ['Continuous'])

Expectation:
I was expecting it to also show me the SingleFrame as an option

medasani456, 2018-04-11:
Did you checked weather camera is having Single frame mode?
I worked with TIS camera having only "Continuous Grab", but not Single grab

EmmanuelP, 2018-04-12:
What it the output of `arv-tool-0.6 features` ?

EmmanuelP, 2018-05-24:
No update, I'm closing this issue.

Cheers.

closed 2018-05-24 by EmmanuelP