AravisProject/aravis #147: VRMagic LineCam3d: integer feature mis-interpreted as int64 (is an int32)

eudoxos, 2018-04-13:
Hi, I am gratefully interfacing the LineCam3d sensor (http://www.vrmagic.com/vrmagic-imaging/3d-sensors/linecam3d/) using Aravis. Everything works okay, except that some integer parameters confuse int64/int32, signed/unsigned or byte order (I am not sure).

I queried the parameters using this code (it writes the value, the value interpreted as int32, and the bounds):

```
#include<arv.h>
#include<iostream>
int main(void){
   ArvCamera *camera;
   ArvDevice *device;
   camera=arv_camera_new (NULL);
   device=arv_camera_get_device(camera);
   for(const auto& f:{"Scan3dCoordinateAScale_Numerator","Scan3dCoordinateAScale_Denominator","Scan3dCoordinateAOffset_Numerator","Scan3dCoordinateAOffset_Denominator","Scan3dCoordinateCScale_Numerator","Scan3dCoordinateCScale_Denominator","Scan3dCoordinateCOffset_Numerator","Scan3dCoordinateCOffset_Denominator"}){
      gint64 v=arv_device_get_integer_feature_value(device,f);
      gint64 lo,hi;
      arv_device_get_integer_feature_bounds(device,f,&lo,&hi);
      std::cout<<f<<": "<<v<<" ["<<*(gint32*)(&v)<<"] ("<<lo<<"..."<<hi<<")"<<std::endl;
   }
   g clear_object(&camera);
   return 0;
}
```
and this is the result:
```
Scan3dCoordinateAScale_Numerator: 166 [166] (-2147483648...2147483647)
Scan3dCoordinateAScale_Denominator: 65534 [65534] (1...2147483647)
Scan3dCoordinateAOffset_Numerator: 4294967294 [-2] (-2147483648...2147483647)
Scan3dCoordinateAOffset_Denominator: 2 [2] (1...2147483647)
Scan3dCoordinateCScale_Numerator: 108 [108] (-2147483648...2147483647)
Scan3dCoordinateCScale_Denominator: 65534 [65534] (1...2147483647)
Scan3dCoordinateCOffset_Numerator: 0 [0] (-2147483648...2147483647)
Scan3dCoordinateCOffset_Denominator: 2 [2] (1...2147483647)
```

The issue is with the 2nd line (``Scan3dCoordinateAOffset_Numerator``) being really negative (-2 is the correct value but it read as bogus `4294967294`) . The bogus value is displayed by other Aravis-based codes (such as ``qavr``), but VRMagic-provided VRmGevCamLab interprets the value correctly:

[image]

This is the description in the GenICam XML downloaded from the camera:

```
  <Integer Name="Scan3dCoordinateAOffset_Numerator">
    <ToolTip>Numerator of Offset for the lateral A-Coordinate</ToolTip>
    <Description>Numerator of Offset for the lateral A-Coordinate</Description>
    <DisplayName>Coordinate A Offset Numerator</DisplayName>
    <Streamable>Yes</Streamable>
    <pValue>Scan3dCoordinateAOffset_NumeratorReg</pValue>
    <Min>-2147483648</Min>
    <Max>2147483647</Max>
    <Representation>Linear</Representation>
  </Integer>
```

I would like to discover where is the problem: is it an issue between the keyboard and the chair? Bug in VRMagic's firmware? In Aravis? Is any integer value possibly affected by this? I could dump the communication with Wireshark and send you privately, if you want to look at it closer (I don't have the ability to do so).

EmmanuelP, 2018-04-13:
What is the definition of Scan3dCoordinateAOffset_NumeratorReg in the Genicam XML data ?

EmmanuelP, 2018-04-16:
Please try this branch: https://github.com/AravisProject/aravis/tree/wip/signedness

eudoxos, 2018-04-16:
Hi Emmanuel, thanks for the effort, but so far same result. I think it is not just `<Sign>` (which is definitely a worthy fix in itself) but also `<Length>` - it should be 4 but guint64 uses 8 bytes. So (in little endian) `FE FF FF FF` gets interpreted as :

* -2 as int16
* 65534 as uint16
* -2 as int32 (this would be the correct one)
* 4294967294 as uint32 and also as (with trailing `00 00 00 00`) as int64 and uint64.

EmmanuelP, 2018-04-17:
Could you please rerun your test code with ARV_DEBUG set to genicam:3 (`export ARV_DEBUG=genicam:3`) ?

eudoxos, 2018-04-17:
```
[Gc::register_feature_node] Register node 'AOIColumnOffset' [Integer]
[Gc::register_feature_node] Register node 'AOIColumnOffsetReg' [IntReg]
[Gc::register_feature_node] Register node 'AOIHeight' [Integer]
[Gc::register_feature_node] Register node 'AOIHeightReg' [IntReg]
[Gc::register_feature_node] Register node 'AOIRowOffset' [Integer]
[Gc::register_feature_node] Register node 'AOIRowOffsetReg' [IntReg]
[Gc::register_feature_node] Register node 'AOIWidth' [Integer]
[Gc::register_feature_node] Register node 'AOIWidthReg' [IntReg]
[Gc::register_feature_node] Register node 'AcqEnabledReg' [IntReg]
[Gc::register_feature_node] Register node 'AcqModeReg' [IntReg]
[Gc::register_feature_node] Register node 'AcquisitionControl' [Category]
[Gc::register_feature_node] Register node 'AcquisitionFrameRateEnable' [Boolean]
[Gc::register_feature_node] Register node 'AcquisitionFrameRateEnableReg' [IntReg]
[Gc::register_feature_node] Register node 'AcquisitionFrameRateMax_mHz' [Integer]
[Gc::register_feature_node] Register node 'AcquisitionFrameRateMax_mHzReg' [IntReg]
[Gc::register_feature_node] Register node 'AcquisitionFrameRate_mHz' [Integer]
[Gc::register_feature_node] Register node 'AcquisitionFrameRate_mHzReg' [IntReg]
[Gc::register_feature_node] Register node 'AcquisitionMode' [Enumeration]
[Gc::register_feature_node] Register node 'AcquisitionStart' [Command]
[Gc::register_feature_node] Register node 'AcquisitionStop' [Command]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeMax' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeMaxReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeMin' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeMinReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeScale' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateARangeScaleReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeMax' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeMaxReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeMin' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeMinReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeScale' [Integer]
[Gc::register_feature_node] Register node 'ActCalibCurrentCoordinateCRangeScaleReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomLeftX' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomLeftXReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomLeftY' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomLeftYReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomRightX' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomRightXReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomRightY' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainBottomRightYReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainFirstColumn' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainFirstColumnReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainFirstRow' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainFirstRowReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainLastColumn' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainLastColumnReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainLastRow' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainLastRowReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainTopLeftX' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainTopLeftXReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainTopLeftY' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainTopLeftYReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainTopRightX' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainTopRightXReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibDomainTopRightY' [Integer]
[Gc::register_feature_node] Register node 'ActCalibDomainTopRightYReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeMax' [Integer]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeMaxReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeMin' [Integer]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeMinReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeScale' [Integer]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateARangeScaleReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateCRangeMax' [Integer]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateCRangeMaxReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateCRangeMin' [Integer]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateCRangeMinReg' [IntReg]
[Gc::register_feature_node] Register node 'ActCalibFullCoordinateCRangeScale' [Integer]
[... cut ...]
[GcRegisterNode::_get_integer_value] address = 0xc0ec, value = 0x6c
Scan3dCoordinateCScale_Numerator: 108 [108] (-2147483648...2147483647)
[GcRegisterNode::_get_integer_value] address = 0xc0f0, value = 0xfffe
Scan3dCoordinateCScale_Denominator: 65534 [65534] (1...2147483647)
[GcRegisterNode::_get_integer_value] address = 0xc0fc, value = 0x0
Scan3dCoordinateCOffset_Numerator: 0 [0] (-2147483648...2147483647)
[GcRegisterNode::_get_integer_value] address = 0xc100, value = 0x2
Scan3dCoordinateCOffset_Denominator: 2 [2] (1...2147483647)
```

EmmanuelP, 2018-04-17:
Ok, I have missed the standard not masked case. Please try the updated wip/signedness branch.

eudoxos, 2018-04-17:
Almost. You need to fix parentheses in one of the conditions, then it works perfectly (sorry, don't know how to do a quick pull request): `((value & (1 << (length * 8 - 1))) != 0)`

EmmanuelP, 2018-04-17:
Good catch. I have pushed a fix  and the related unit test.

EmmanuelP, 2018-04-18:
Everything is on master now.

closed 2018-04-18 by EmmanuelP