AravisProject/aravis #149: Error on opening Schema

ArkadiuszRaj, 2018-04-19:
I was developing my application for some time, using some rpevious version of Aravis. But occasionally I am hit by:

`*** Error in./app': double free or corruption (fasttop): 0x0000007f7cc00940 ***`

Also after 10-15min it was not possible to retrieve images. I mean 
`arv_buffer = arv_stream_timeout_pop_buffer (status.stream, 200000); `
continously time outs.

To isolate problem I have started my app nbut with fake source (no aravis). Worked fine for more that 24h. Thus I have decided to try latest revision of aravis lib taken from master branch (hour ago) -> commit 7bc085972f3eaed7ed9698bae8ce3a6fe7c491e0

And to my surprise I can not start my app. arv-tool works but arv-test does not:
[genicam.xml.zip](https://github.com/AravisProject/aravis/files/1928669/genicam.xml.zip)

```
># arv-test-0.6 
Testing device 'Chameleon3 CM3-U3-31S4M' from 'Point Grey Research'

(process:10949): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed

(process:10949): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
Genicam: FAILED Invalid Genicam XML data
Sensor size read: 2048 1536
```

arv-tool-0.6 features works smoothly. However in my application aravise raises SIGSEGV, here comes excerpt from gdb

```
[Switching to Thread 0x7fb29530f0 (LWP 10986)]
0x0000007fb7f23604 in evaluate (token_stack=<optimized out>, variables=0x7fb4aa0790, v_int64=v_int64@entry=0x0, v_double=0x7fac004ca8, v_double@entry=0x7fb29521c0) at arvevaluator.c:953
[genicam.xml.zip](https://github.com/AravisProject/aravis/files/1928649/genicam.xml.zip)

953			*v_double = arv_value_get_double (&stack[0].value);
(gdb) bt
#0  0x0000007fb7f23604 in evaluate (token_stack=<optimized out>, variables=0x7fb4aa0790, v_int64=v_int64@entry=0x0, v_double=0x7fac004ca8, v_double@entry=0x7fb29521c0) at arvevaluator.c:953
#1  0x0000007fb7f25918 in arv_evaluator_evaluate_as_double (evaluator=0x7fac0afab0, error=error@entry=0x0) at arvevaluator.c:1222
#2  0x0000007fb7f3bd04 in _get_node_integer_value (gc_integer=<optimized out>, node_type=ARV_GC_CONVERTER_NODE_TYPE_VALUE, error=0x7fb2952240) at arvgcconverter.c:478
#3  0x0000007fb7f3a8ac in _update_variables (gc_swiss_knife=gc_swiss_knife@entry=0x7fac08f550, error=0x7fb2952280, error@entry=0x7fb29522a0) at arvgcswissknife.c:215
#4  0x0000007fb7f3ab0c in arv_gc_swiss_knife_get_float_value (gc_float=<optimized out>, error=0x7fb29522d0) at arvgcswissknife.c:277
#5  0x0000007fb7f30a20 in arv_gc_property_node_get_double (node=<optimized out>, error=0x7fb2952310, error@entry=0x7fb2952330) at arvgcpropertynode.c:407
#6  0x0000007fb7f370e8 in arv_gc_float_node_get_min (gc_float=<optimized out>, error=0x7fb2952370) at arvgcfloatnode.c:257
#7  0x0000007fb7f40abc in arv_device_get_float_feature_bounds (device=0x7fac0158b0, feature=<optimized out>, min=min@entry=0x7fac000938, max=max@entry=0x7fac000940) at arvdevice.c:622
#8  0x0000007fb7f2c98c in arv_camera_get_exposure_time_bounds (camera=0x7fac1f7880, min=0x7fac000938, max=0x7fac000940) at arvcamera.c:1339
```

I am attaching xml file taken from camera.

EmmanuelP, 2018-04-23:
> And to my surprise I can not start my app. arv-tool works but arv-test does not
arv-test is a work in progress.  Warnings are expected.

I'm experiencing the same crash, but only when I build aravis using meson. May be it comes from a compiler flag. Could you attach the output of `make V=1` ?

ArkadiuszRaj, 2018-04-23:
Attached log from configuring & compiling.

BTW, problem with timeouts was related to low quality usb3 connection.
[aravis-build-log.txt](https://github.com/AravisProject/aravis/files/1937308/aravis-build-log.txt)

EmmanuelP, 2018-04-23:
I have pushed a fix for a memory corruption in the evaluator. Could you try current master please ?

EmmanuelP, 2018-04-23:
I it doesn't fix the crash, could you give the output of your application with `ARV_DEBUG` set to `evaluator:3` ?

ArkadiuszRaj, 2018-04-23:
Yes, compiles well, works well.
THX!

EmmanuelP, 2018-04-23:
Ok. Thanks for the report.

closed 2018-04-23 by EmmanuelP