AravisProject/aravis #150: Problem with aravis

larsrindorf, 2018-04-30:
Hi

When I run the tests (cd tests/, make check) the evaluator fails. The test_suite.log is attached.
[test-suite.log](https://github.com/AravisProject/aravis/files/1960526/test-suite.log)

../test-driver: line 107: 26540 Aborted                 (core dumped) "$@" > $log_file 2>&1
FAIL: evaluator
PASS: buffer
PASS: misc
PASS: genicam
PASS: fake
PASS: fakegv
============================================================================
Testsuite summary for aravis 0.5.11
============================================================================
# TOTAL: 6
# PASS:  5
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

Any ideas how to troubleshoot that? I tried aravis 0.6 (master branch) as well with the same error.

EmmanuelP, 2018-04-30:
Already fixed by c1d4a267a0ac688e3e9b8d2651754c203427821f

EmmanuelP, 2018-04-30:
May be not, I did not check the log file, sorry.

EmmanuelP, 2018-04-30:
What platform do you use ?

larsrindorf, 2018-04-30:
Hi Emmanuel

I'm on Ubuntu 16 all is updated.  

Thanks

EmmanuelP, 2018-04-30:
On x86_64 ?

EmmanuelP, 2018-04-30:
Also, could you attach the output of `ARV_DEBUG=evaluator:3 ./tests/evaluator -p /evaluator/abs64bits` please ?

larsrindorf, 2018-04-30:
Latitude-D630:~/aravis$ ARV_DEBUG=evaluator:3 ./tests/evaluator -p /evaluator/abs64bits
/evaluator/abs64bits: [Evaluator::set_double_variable] PI = 3.14159
[Evaluator::set_double_variable] E = 2.71828
[Evaluator::evaluate_as_int64] Expression = 'ABS(-10000000000)'
[Evaluator::parse_expression] ABS(-10000000000)
[Evaluator::parse_expression] Found 5 items in expression
[Evaluator::parse_expression] 2 items in garbage list
[Evaluator::parse_expression] 3 items in token list
[Evaluator::evaluate_as_int64] Parsing status = 0
(int64) 10000000000
(operator) minus
[Evaluator::set_error] Error 'invalid double function'
[Evaluator::evaluate_as_double] Expression = 'ABS(-10000000000)'
(int64) 10000000000
(operator) minus
(operator) abs
[Evaluator::evaluate] Result = (int64) 1410065408
**
ERROR:evaluator.c:129:expression_test: assertion failed (v_double == data->result_double): (1.41006541e+09 == 1e+10)
Aborted (core dumped)

larsrindorf, 2018-04-30:
Yes x86_64.

larsrindorf, 2018-04-30:
Yes x86_64.

I hope you have any clues about what's going on :-)

larsrindorf, 2018-05-01:
Now it passes!

PASS: evaluator
PASS: buffer
PASS: misc
PASS: genicam
PASS: fake
PASS: fakegv
============================================================================
Testsuite summary for aravis 0.5.12
============================================================================
# TOTAL: 6
# PASS:  6
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================

EmmanuelP, 2018-05-01:
Great. Thanks for the bug report.

Cheers.

closed 2018-05-01 by EmmanuelP