AravisProject/aravis #151: Only 2000 frames from camera

YarochkinMichael, 2018-05-02:
I am running MatrixVision BC-X104eC (firmware 2.24.975.0) under Aravis 0.6 and receiving only 2000 frames after starting continuous acquisition.
And then it is stopping: my callback function is not called anymore.

Should i "release" buffer after processing? But i don`t see arv_buffer_release() function.

EmmanuelP, 2018-05-02:
Sometimes there is a feature that limits the number of acquired frames during a continuous acquisition.

Could you attach the output of `arv-tool-0.6 features` please ?

YarochkinMichael, 2018-05-02:
Here it is
[features.txt](https://github.com/AravisProject/aravis/files/1966724/features.txt)

YarochkinMichael, 2018-05-02:
If i run camera with the same settings using MatrixVision tool (wxPropView), i do not have any limit.

YarochkinMichael, 2018-05-02:
Ok. Some extra input:
-it is 1000: 1000 ARV_STREAM_CALLBACK_TYPE_START_BUFFER and 1000 ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE

- i have 1000 buffers in my app: 
         for (int i = 0; i < this->1000; i++)
            arv_stream_push_buffer(arvStream, arv_buffer_new(this->payload, NULL));

What happens when all the buffers are used?
Should i push it back after usage?
It is wrap around? Should i add buffer again?

In 0.4 it was not an issue.

PS; it is not really my app. done by another developer. i am busy with migration from 0.4 to 0.6

EmmanuelP, 2018-05-02:
While receiving the buffers, each call to arv_pop_buffer must be followed by a call to arv_push_buffer when you are done with the data in the buffer.

Please have a look at tests/arvcameratest.c

YarochkinMichael, 2018-05-02:
Yes. Fixed. Thank you.

One strange thing: this was working in 0.4 version.
Possible?

EmmanuelP, 2018-05-02:
> Possible?

Unlikely. This behaviour has not changed since 0.2.x.

closed 2018-05-02 by EmmanuelP

EmmanuelP, 2018-06-08:
Don't delete the buffer and push it back to the stream input queue.

EmmanuelP, 2018-06-08:
It should be:
```
void arvStreamCallback(void *user_data, ArvStreamCallbackType type, ArvBuffer *buffer)
{
  if (type == ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE) {
    //buffer processing
    arv_stream_push_buffer(buffer);
  }
}
```

YarochkinMichael, 2018-06-08:
Yes of course!
I forgot about ARV_STREAM_CALLBACK_TYPE_START_BUFFER,ARV_STREAM_CALLBACK_TYPE_INIT ....
Thank you!