AravisProject/aravis #152: Man pages fails with Meson

rffontenelle, 2018-05-03:
In Aravis 0.5.12, using Meson and Ninja will fail to build & install the man pages with the error message below. See attached file [build-meson.txt](https://github.com/AravisProject/aravis/files/1971296/build-meson.txt) for full build output.

```
Building documentation for aravis
./aravis-sections.txt:4: warning: No declaration found for ARAVIS_CHECK_VERSION.
./aravis-sections.txt:5: warning: No declaration found for ARAVIS_MAJOR_VERSION.
./aravis-sections.txt:6: warning: No declaration found for ARAVIS_MICRO_VERSION.
./aravis-sections.txt:7: warning: No declaration found for ARAVIS_MINOR_VERSION.
./aravis-sections.txt:8: warning: No declaration found for ARAVIS_HAS_PACKET_SOCKET.
./aravis-sections.txt:9: warning: No declaration found for ARAVIS_HAS_USB.
See aravis-undeclared.txt for the list of undeclared symbols.

warning: failed to load external entity "../../../../tests/arvexample.c"
warning: failed to load external entity "../../../../tests/arvchunkparsertest.c"
```

Environment:
* distro: Arch Linux 64-bit
* packages:
  * aravis = 0.5.12
  * meson = 0.46.0
  * ninja = 1.8.2

EmmanuelP, 2018-05-04:
Did you try to build aravis using a clean checkout ? Because if you run the autotools build, it will leave some files around that breaks the meson build.

rffontenelle, 2018-05-04:
Yes, I always clean the build directory. I use Arch tools for clean chroot build, which install the dependencies, extract the source code and build in a chroot environment.

Not sure it helps, but I read in the generated `ninja.build` and found that the found that the error appears after this command:

> /usr/bin/python /usr/bin/meson --internal commandrunner /build/aravis/src/aravis-ARAVIS_0_5_12 /build/aravis/src/build docs/reference/aravis /usr/bin/python /usr/bin/meson /usr/bin/python /usr/bin/meson --internal gtkdoc --sourcedir=/build/aravis/src/aravis-ARAVIS_0_5_12 --builddir=/build/aravis/src/build --subdir=docs/reference/aravis --headerdirs=/build/aravis/src/aravis-ARAVIS_0_5_12/src --mainfile=aravis-docs.xml --modulename=aravis --mode=auto '--scanargs=--ignore-headers=arvenumtypes.h arvconfig.h arvbufferprivate.h arvchunkparserprivate.h arvinterfaceprivate.h arvdeviceprivate.h arvstreamprivate.h arvgvstreamprivate.h arvgvinterfaceprivate.h arvgvdeviceprivate.h arvvvstreamprivate.h arvuvinterfaceprivate.h arvuvdeviceprivate.h arvuvstreamprivate.h arvfakeinterfaceprivate.h arvfakedeviceprivate.h arvfakestreamprivate.h arvrealtimeprivate.h arv.h@@--rebuild-types' --gobjects-types-file=aravis.types --content-files=/build/aravis/src/aravis-ARAVIS_0_5_12/docs/reference/aravis/aravis-overview.xml@@/build/aravis/src/aravis-ARAVIS_0_5_12/docs/reference/aravis/aravis-gv.xml --installdir=aravis-0.6 '--cflags=-I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -pthread -I/usr/include/libxml2 -I/usr/include/libusb-1.0 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -D_FORTIFY_SOURCE=2' '--ldflags=-L/build/aravis/src/build/src -Wl,-rpath,/build/aravis/src/build/src -laravis-0.6 -lglib-2.0 -lgobject-2.0 -lgio-2.0 -lxml2 -lz -lm -lcap-ng -laudit -lusb-1.0 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now' --cc=cc --ld=cc

EmmanuelP, 2018-05-04:
Perhaps it is related to the meson version. I'm still with meson 0.45.1. I will retry when last meson will be available in fedora 28.

EmmanuelP, 2018-05-27:
'failed to load external entity' errors should be fixed in master now.
I don't get the 'declaration' errors with meson 0.46.1 though.

rffontenelle, 2018-06-07:
I confirm the above 'failed to load external entity' errors is gone. I still get the 'declaration' errors, but at least the documentation is installed except for the "undeclared" ones.

I'm using gtk-doc 1.28.1; what is your gtk-doc version?

EmmanuelP, 2018-08-10:
Missing declarations are fixed by 76e9d11042702e2871596abb5d894874a986462d.

Thanks for the report.

Cheers.

closed 2018-08-10 by EmmanuelP