AravisProject/aravis #154: Aravis - driver issues

larsrindorf, 2018-05-04:
Aravis cannot find my USB3 camera even though it is visible with

mplayer tv:// -tv driver=v4l2:device=/dev/video0

an aravis test yields:
:~/aravis/tests$ ./arv-camera-test 
Looking for the first available camera
No camera found

How do I set the driver and the device in Aravis?

Best, Lars

EmmanuelP, 2018-05-04:
Hi Lars,

What is the model and manufacturer of your camera ?
Aravis only supports USB3Vision cameras. Is this the case for your device ?

larsrindorf, 2018-05-04:
Hi Emmanuel 

It is imagingsource and it is USB3.0. 

I suspect the driver or codec needs specifying somewhere if this makes sense...

EmmanuelP, 2018-05-04:
Probably a permission access issue.
Did you read https://github.com/AravisProject/aravis#usb-permissions ?

larsrindorf, 2018-05-04:
The aravis.rules is in the /etc/udev/rules.d/ folder and the imagingsource camera is there too
# The Imaging Source
SUBSYSTEM=="usb", ATTRS{idVendor}=="199e", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"

EmmanuelP, 2018-05-04:
What is the output of: `lsusb` ?
The output of `arv-tool-0.6 -d all:3` ? The output of the same command, but as root ?

larsrindorf, 2018-05-04:
The camera can be viewed using mpplayer:
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0a2a Intel Corp. 
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 1bcf:28c1 Sunplus Innovation Technology Inc. 
Bus 001 Device 007: ID 199e:8706 The Imaging Source Europe GmbH 
Bus 001 Device 006: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

 arv-tool-0.6 -d all:3
Found 0 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 10.0.0.118
No device found

sudo arv-tool-0.6 -d all:3
[sudo] password for lars: 
Found 0 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 10.0.0.118
No device found

larsrindorf, 2018-05-04:
Another information is that the camera is gray scale (GREY)

EmmanuelP, 2018-05-04:
Could you attach the output of `sudo lsusb -v -d 199e:` please ?

> The camera can be viewed using mpplayer
mplayer uses the v4l2 interface, which is completely different than USB3Vision and uses different permissions.

larsrindorf, 2018-05-09:
Here is the output:
[aravis_usb.txt](https://github.com/AravisProject/aravis/files/1987729/aravis_usb.txt)

EmmanuelP, 2018-05-10:
It looks like your camera does not implement the USB3Vision interface. There should be several interface descriptors with bInterfaceClass == 239 in the lsusb output.
Aravis only implements GigEVision and USB3Vision for now.

closed 2018-05-10 by EmmanuelP

larsrindorf, 2018-05-10:
Thanks Emmanuel for tracking down root cause!