AravisProject/aravis #159: arv_gv_device_new() multiple devices

mehtajaghvi, 2018-05-17:
1. Function used
arv_gv_device_new (GInetAddress *interface_address, GInetAddress *device_address)
2. what is happening
 Multiple cameras are calling this function in parallel
3. Error
(process:24): Aravis-CRITICAL **: arv_dom_node_append_child: self is not a ArvDomNode
(process:24): Aravis-CRITICAL **: arv_dom_node_append_child: self is not a ArvDomNode
process:24): Aravis-CRITICAL **: arv_dom_node_get_parent_node: assertion 'ARV_IS_DOM_NODE (self)' failed

EmmanuelP, 2018-05-17:
Thanks for the report.

Could you attach a simple example that shows this issue ?

mehtajaghvi, 2018-05-21:
1. Trying to capture images using the camer-ip instead of the serial number.
2. Calling the function arv_gv_device_new in 2 separate threads for multiple cameras.
3. The system runs ok, when we  have one camera. But whenever we have multiple cameras calling the function, the error is generated and does not create a camera instance. 
3. Code for creating the camera object

void Camera::camera_new_ip()
{
    std::cout << "creating device by IP" << std::endl;

    GInetAddress *host_address = g_inet_address_new_from_string(_camera.hostIP);
    GInetAddress *device_address = g_inet_address_new_from_string(_camera.camIP);

    _camera.device = arv_gv_device_new(host_address, device_address);

    std::cout << "created....... device by IP" << std::endl;

    if (!ARV_IS_DEVICE(_camera.device))
    {

        const std::string &what = "Could not create camera";
        std::cout << "could not create....... device by IP" << std::endl;
        throw std::runtime_error(what);
    }

    _camera.camera = (ArvCamera *)g_object_new(ARV_TYPE_CAMERA, "device", _camera.device, NULL);

     
}

EmmanuelP, 2018-05-24:
By simple example I mean something in a single source file that exhibits this issue, I can compile and run.

azmosavi, 2018-05-24:
@mehtajaghvi I exactly have the same problem as you mentioned clearly. Did you find any solution for that.

EmmanuelP, 2018-05-30:
@mehtajaghvi , could you try to put a mutex around the arv_gv_device_new calls ?

```
g_mutex_lock (&mutex)
arv_gv_device_new (...)
g_mutex_unlock (&mutex)
```

with mutex defined globally using `static GMutex mutex`

And tell us if it fixes your issue.

EmmanuelP, 2018-08-10:
Did the mutex suggestion fix your issue ?

If not, please reopen this issue.

Cheers.

closed 2018-08-10 by EmmanuelP