AravisProject/aravis #160: Byte order issue GevSCDA = No image stream

meuchel, 2018-05-19:
I have strange issues getting the image stream when using the aravis Library with a Flir camera on an ubuntu 16.04 system. I'm using aravis-0.5.12 (same happens with aravis 0.4)

With the Pleora eBus Player on the same system I can get the image stream but not with the Aravis Library.
It seems that the byte order in GevSCDA is wrong (?) (see below the parameter output from arv-tool)
How can this happen, what I'm doing wrong?
I have also another Flir camera of different type and with this camera the stream is working (with eBus Player and Aravis Library)

Here the parameters when the applications are running.

BAD with Aravis Lib:
GevGVSPExtendedIDMode = Off
GevSCPSFireTestPacket = false
GevSCPSDoNotFragment = true
GevSCPSBigEndian = false
GevSCPSPacketSize = 64 (min:46;max:16266)
GevSCDA = 3691096256   -> DC01A8C0 -> 220.1.168.192

GOOD with Pleora eBus Player:
GevGVSPExtendedIDMode = Off
GevSCPSFireTestPacket = false
GevSCPSDoNotFragment = true
GevSCPSBigEndian = false
GevSCPSPacketSize = 8164 (min:46;max:16266)
GevSCDA = 3232235996 -> C0A801DC -> 192.168.1.220

EmmanuelP, 2018-05-24:
Hi,

Could you attach the relevant part of genicam data related to GevSCDA ?  (GevSCDA and GevSCDAReg nodes)

Genicam data can be obtained using `arv-tool-0.6 genicam`.

EmmanuelP, 2018-08-10:
No answer. Please reopen if you have the requested informations.

Cheers.

closed 2018-08-10 by EmmanuelP