AravisProject/aravis #162: use of arv_update_device_list

manuelvolk, 2018-05-22:
I have a short question and I'm not sure it is a bug, wanted behavior or a misuse of the Aravis lib.

I try to update the device list all 10 seconds to check if new cameras are available and could be used by my application. So far so good and everything works like as expected (when I connect a new camera it is listed as available). 

But when I disconnect some devices which are already been found the devices are be still listed by the
application. My expected behavior was that this is not the case. 
 
```
#include <string>
#include <iostream>

#include "arv.h"
#include "glib.h"

struct Global
{
  GMainLoop* main_loop;
  gboolean bCancel;
} global;

static void set_cancel(int signal) { global.bCancel = TRUE; }

// PeriodicTask_callback()
static gboolean PeriodicTask_callback(void* )
{
  if (global.bCancel)
  {
    g_main_loop_quit(global.main_loop);
    return FALSE;
  }

  uint64_t nDevices=0;
  Print out some useful info.
  std::cout<<"Update Device List:"<<std::endl;
  arv_update_device_list();
  nDevices = arv_get_n_devices();
  std::cout<<<"# Number of found Devices: "  + std::to_string(nDevices) << std::endl;

  for (int i = 0; i < nDevices; i++)
  {
    std::string device_ID = arv_get_device_id(i);
    std::cout<<device_ID<<std::endl;
  }

  return TRUE;
} // PeriodicTask_callback()

int main(int argc, char** argv)
{
  global.bCancel = false;

  g_timeout_add_seconds(10, PeriodicTask_callback, NULL);

  void (*pSigintHandlerOld)(int);
  pSigintHandlerOld = signal(SIGINT, set_cancel);

  global.main_loop = 0;
  global.main_loop = g_main_loop_new(NULL, FALSE);
  g_main_loop_run(global.main_loop);

  signal(SIGINT, pSigintHandlerOld);

  g_main_loop_unref(global.main_loop);

  arv_shutdown();

  return 0;
} // main() `
```

manuelvolk, 2018-05-23:
Hi Emmanuel,
yes I mean physically disconnect the device. 
So, therefore it is a bug?

EmmanuelP, 2018-05-23:
I am using a slightly modified version of your test pushed here: https://github.com/AravisProject/aravis/tree/wip/device-scan-test (see tests/arvdevicescantest.c)

I am not able to replicate your issue.

Could you please rerun your test with `ARV_DEBUG` set to `all:3`, and attach the output here ?

manuelvolk, 2018-05-25:
Hi I test it again with Debug on, here are the results:

[ INFO] [1527260642.291269798]: Update Device List:
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface *.*.*.*
[ INFO] [1527260643.292986722]: # Number of found Devices: 0
[ INFO] [1527260653.675749665]: Update Device List:
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface *.*.*.*
[ INFO] [1527260654.677805399]: # Number of found Devices: 0
[ INFO] [1527260663.675669226]: Update Device List:
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface *.*.*.*
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = *
name         = 
model        = *
address      = 192.168.*.*
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
""
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device '****-41******5' found (interface 192.168.*.*) user_name '' - MAC_name '00:1b:**:20:**:**'
[ INFO] [1527260664.793018560]: # Number of found Devices: 1
[ INFO] [1527260673.676030406]: Update Device List:
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface *.*.*.*
[ INFO] [1527260674.678048437]: # Number of found Devices: 1
[ INFO] [1527260683.675399679]: Update Device List:
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.*.*
[GvDiscoverSocket::new] Add interface *.*.*.*
[ INFO] [1527260684.677639014]: # Number of found Devices: 1

I use start to hide some sensitive information!

As you can see I start the test code with no camera connected. After that I connect the camera and it is found like expected. Then I disconnect the camera again, but the device count is still one.

EmmanuelP, 2018-05-26:
Did you try with arv-viewer ? The list of available device should be updated each time you click on the refresh button.

EmmanuelP, 2018-05-30:
> When I install version 0.4.1 I can use arv-tool arv-viewer out of the box but when I
install 0.5.12 I have to first add to the ld_library_path /usr/lib/local.

I don't think the installation of arv-viewer has changed between 0.4.x and 0.5.x. Did you install 0.4.1 in /usr/local, or /usr ?

> when I physically disconnect a camera it takes some time until the control-lost callback
is called. Is there a way to call the callback directly after the connection loss?

With the GigEVision protocol or USB3Vision ? Using GigEVision, there is no way to know when the camera is disconnected except when one of the exchange reach a timeout.

manuelvolk, 2018-05-30:
We use a GigEVision camera:
Ok, I read something about heartbeat support. Is it not possible to check this heartbeat also with the GigEVision protocol?

EmmanuelP, 2018-05-31:
> Ok, I read something about heartbeat support. Is it not possible to check this heartbeat also with the GigEVision protocol?

That is what aravis does. Heartbeat register is read every 5s. There is a compile time option (fast-heartbeat) that increase the read rate.

closed 2018-06-07 by manuelvolk