AravisProject/aravis #163: Aravis typelib not being installed

ChaosCA, 2018-05-23:
I'm having issues installing on Ubuntu 16.04. I was able to build with the following commands:
```
./configure --enable-viewer --enable-gst-plugin --enable-introspection=yes
make
sudo make install
sudo ldconfig
```

All of which resolved without issues. However no typelib file was created(as it should have been) inside /usr/lib/girepository-1.0.

I need the typelib for Aravis python bindings..

Thanks

ChaosCA, 2018-05-23:
After a successful install I now get errors when trying to `make`.

```
arvuvcp.c: In function ‘arv_uvcp_packet_type_to_string’:
arvuvcp.c:127:29: error: ‘ARV_TYPE_UVCP_PACKET_TYPE’ undeclared (first use in this function)
  return arv_enum_to_string (ARV_TYPE_UVCP_PACKET_TYPE, value);
                             ^
arvuvcp.c:127:29: note: each undeclared identifier is reported only once for each function it appears in
arvuvcp.c: In function ‘arv_uvcp_command_to_string’:
arvuvcp.c:133:29: error: ‘ARV_TYPE_UVCP_COMMAND’ undeclared (first use in this function)
  return arv_enum_to_string (ARV_TYPE_UVCP_COMMAND, value);
                             ^
arvuvcp.c: In function ‘arv_uvcp_packet_type_to_string’:
arvuvcp.c:128:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
arvuvcp.c: In function ‘arv_uvcp_command_to_string’:
arvuvcp.c:134:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:913: recipe for target 'arvuvcp.lo' failed
```

EmmanuelP, 2018-05-24:
Hi,
> After the successful install 

Successful install of what ?

Which version of aravis do you try to install ? Please use the tarballs hosted here: http://ftp.acc.umu.se/pub/GNOME/sources/aravis/

If the issue persists, please attach the complete install log here.

ChaosCA, 2018-05-24:
This project. I used the latest tarball and ran the commands I gave above. The first time I tried it seemed to complete fine but now I get the error I posted above.

EmmanuelP, 2018-05-27:
Sorry, but you did not answer to my question: which version of aravis are you using ?
Did you try to ip your installation, and restart ?

closed 2018-05-28 by ChaosCA

arbustoadamo, 2019-06-02:
@ChaosCA  How did you get to fix the error? The same thing has happened to me since 3 days ago.