AravisProject/aravis #166: Disable USB Auto Capturing

ChaosCA, 2018-05-29:
Currently Aravis will automatically latch onto USB cameras it can. This creates an issue for some cameras where I use the manufacturer API as their API can't get access from Aravis.

Is there a way to prevent Aravis from automatically capturing cameras?

EmmanuelP, 2018-05-30:
The USB device is just kept open by aravis, because it would not be good performance wise to open/close it at each access, especially during the streaming.

What could be done would be to add a libusb_device_handle accessor. What does the vendor API look lilke ? Is there a way to give it an already open USB device handle ?

ChaosCA, 2018-05-30:
Even recompiling with USB Support disabled in aravis hasn't seemed to fix the issue..

ChaosCA, 2018-05-30:
I've sent a message to Point Grey in that case and will get back to you. It seems to me like it doesn't play nicely with Aravis installed however.

Using multiprocessing to connect to different cameras using their API while Aravis handles GigE cameras has been failing. The USB side of things doesn't seem to allow more than one synchronous connection and I figured aravis might be causing the issue.

ChaosCA, 2018-05-30:
It wasn't Aravis' fault. Sorry for the issue.
Point Grey is looking into it.

Thanks @EmmanuelP

closed 2018-05-30 by ChaosCA