AravisProject/aravis #167: Cannot open the camera

multiflexi, 2018-05-31:
Hi,
I am testing infrared camera Tachyon 16k from NIT. I tried latest Aravis release and also master branch but without success. I am attaching output of arv-test and camera features and screenshots or viewer and QArv. I was able to acquire image in 3rd party viewer. I think the issue is that Aravis does not figure out the resolution which is 128x128 and shows just 0 in arv-test. Any ideas?

[arv-test.log](https://github.com/AravisProject/aravis/files/2057870/arv-test.log)
[cam_features.txt](https://github.com/AravisProject/aravis/files/2057871/cam_features.txt)
[image]
[image]

EmmanuelP, 2018-06-01:
Please attach the output of `arv-camera-test -d all:3`.

multiflexi, 2018-06-04:
I let it run for 30 min, I hope it is enough.
[arv-camera-test.log](https://github.com/AravisProject/aravis/files/2067362/arv-camera-test.log)

multiflexi, 2018-06-05:
Is the issue evident from the log?

EmmanuelP, 2018-06-06:
Nope. The region seems to be correctly returned by arv-camera-test.
Could you attach the output of:
```
arv-tool-0.6 control SensorHeight SensorWidth Width Height -d device:3,evaluator:3
```

multiflexi, 2018-06-06:
[arv-tool.log](https://github.com/AravisProject/aravis/files/2075260/arv-tool.log)
Here you go.

EmmanuelP, 2018-06-06:
And the output of: `arv-tool-0.6 features` ?

multiflexi, 2018-06-06:
[arv-tool-0.6.log](https://github.com/AravisProject/aravis/files/2076336/arv-tool-0.6.log)

multiflexi, 2018-06-06:
I am now investigating what is wrong. Does Aravis require variables SensorWidth and SensorHeight in the XML file? According to Genicam SFNC they are recommended but not mandatory. When I compare the XML from camera and simulator which works, this is the difference I see: the camera  XML is missing those 2 variables and has only WidthMax and HeightMax.

EmmanuelP, 2018-06-06:
Thanks for investigating this issue.
SensorHeight and SensorWidth are not used by the viewer.
I would recommend to try to debug with tests/arv-camera-test first. It is much more simpler to understand than arv-viewer with all that UI related code.
As I said previously, my first bet would be a non standard acquisition setup sequence. In order to set a fixed framerate acquisition, a number of features must be set, and this is most of the time vendor dependent. Have a look at the camera documentation.

EmmanuelP, 2018-06-06:
It is a bit strange there is no TriggerMode and TriggerSelector features.