AravisProject/aravis #168: Timestamp jump in the past

YarochkinMichael, 2018-06-01:
Hello, Emmanuel

I am using MatrixVision camera (BC-x104eC)
Timestamp is included in chunk data. That is how i am extracting it:
currentTimestamp = arv_chunk_parser_get_integer_value(streamAravis->GetParser(), buffer, "ChunkTimestamp");

After changing exposure time of the camera this value is jumping in the past: previousTimestamp < currentTimestamp.
Also it is happening once in a while during acquisition.

Here is log file. There you can see "currentTimestamp" and "previousTimestamp - currentTimestamp"
[timedif 2.txt](https://github.com/AravisProject/aravis/files/2062598/timedif.2.txt)

If i  am using SDK from Mv it is not happening.

Can you help me to find the reason, please?

Mikl

EmmanuelP, 2018-06-01:
Hi Mikl,

Could you try to modify your software and also display the frame id, timestamp and system_timestamp returned by arv_buffer_get_frame_id, arv_buffer_get_timestamp and arv_buffer_get_system_timestamp, and attach the result here ?

YarochkinMichael, 2018-06-04:
Hello, Emmanuel

Sorry. It was bug in our firmware.
Aravis is working fine. Thank you.

Mikl

EmmanuelP, 2018-06-04:
No problem.

Cheers.

closed 2018-06-04 by EmmanuelP