AravisProject/aravis #170: Determining if a buffer is ready

ChaosCA, 2018-06-07:
Hello, I'm using aravis with a GigE camera and I'm having issues with images appearing split in two.

From what I understand this could mean that a frame is being popped before it is ready. I'm using the Python bindings for aravis but the idea is the same.

```
self.set_feature("AcquisitionMode", "Continuous") #no acquisition limits
self.start_acquisition(10)
```

This starts the camera with 10 buffers. Then in a loop I can grab the images:
```
    def try_pop_frame(self, timestamp=False):
        """
        return the oldest frame in the aravis buffer
        """
        buf = self.stream.try_pop_buffer()
        if buf:
            frame = self._array_from_buffer_address(buf)
            self.stream.push_buffer(buf)
            if timestamp:
                return buf.get_timestamp(), frame
            else:
                return frame
        else:
            if timestamp:
                return None, None
            else:
return None
```

However the frame returned(after the initial one) will usually look like so:
[image]

Is there any way to resolve this? I've tried playing around with the PacketSize and SCPD but without success.

Thank you!

EmmanuelP, 2018-06-08:
Hi,

The buffer can not be popped before it is ready, which does not mean it is complete, the acquisition of a frame can be stopped by a timeout if there is missing packets.
You can check a buffer is complete using buffer.get_status.
Also, what may happen is the buffer data are overwritten before it is actually displayed. You should only push the buffer back to stream when you are done with it. 

Cheers.

ChaosCA, 2018-06-08:
Hey Thanks Emmanuel.

I understand you do not support the python bindings officially but is there anything you’d recommend in changing how the buffer is popped then in this?

https://github.com/SintefRaufossManufacturing/python-aravis/blob/master/aravis.py#L170

Specifically in your last sentence you said that the buffer should only be pushed back after you have finished processing the image. I assume this is because it’s a pointer that is getting overwritten? 

The issue only tends to happen if I make multiple requests, not on a single image grab.

EmmanuelP, 2018-06-08:
> I understand you do not support the python bindings officially but is there anything you’d recommend in changing how the buffer is popped then in this?

No, it looks fine.

> However in that code, the image array is copied before it is pushed back. So it’s unlikely the issue is there

Right.
Check for buffer status. Also, you may want to set the ARV_DEBUG environment variable in order to get some debug output:
```
export ARV_DEBUG=stream:2
```

ChaosCA, 2018-06-08:
I've added the debugging output. However I can't check the status of the buffer from within Python. It should be handled from within the Aravis library [here](https://github.com/SintefRaufossManufacturing/python-aravis/blob/master/ctypes-based/aravis.py#L219).

On startup I see the following debug: [GvStream::stream_new] Destination stream port = 48916
[GvStream::stream_new] Source stream port = 48916
[GvStream::loop] Standard socket method

Popping from the buffer 3 times still results in split images without any additional debug information. Then on stop: 
```
  bins  ;Buffer r
       0;       0
-------------
>=   200;       0
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[GvStream::finalize] n_completed_buffers    = 1
[GvStream::finalize] n_failures             = 3
[GvStream::finalize] n_timeouts             = 3
[GvStream::finalize] n_aborteds             = 0
[GvStream::finalize] n_underruns            = 54925
[GvStream::finalize] n_missing_frames       = 3
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 58172
[GvStream::finalize] n_missing_packets      = 3126
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 54925
[GvStream::finalize] n_resend_requests      = 0
[GvStream::finalize] n_resent_packets       = 0
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 0 buffer[s] in input queue
[Stream::finalize] Flush 1 buffer[s] in output queue
```

I only use 1 buffer because I find that using a large buffer means the images don't get updated for a long, long time.
However I see multiple failures.

I then tried adding a small delay of 1 second, so it grabs 3 images in 3 seconds, the output changes to:
```
  bins  ;Buffer r
       0;       0
-------------
>=   200;       0
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[GvStream::finalize] n_completed_buffers    = 2
[GvStream::finalize] n_failures             = 2
[GvStream::finalize] n_timeouts             = 2
[GvStream::finalize] n_aborteds             = 0
[GvStream::finalize] n_underruns            = 78200
[GvStream::finalize] n_missing_frames       = 3
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 81133
[GvStream::finalize] n_missing_packets      = 2084
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 78200
[GvStream::finalize] n_resend_requests      = 75
[GvStream::finalize] n_resent_packets       = 75
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 0 buffer[s] in input queue
[Stream::finalize] Flush 1 buffer[s] in output queue
```

With an FPS of 16, the camera should be able to handle 1 picture a second..

Thank you again @EmmanuelP I appreciate your help

EmmanuelP, 2018-06-08:
> However I can't check the status of the buffer from within Python

I don't understand. Why can't you call arv_buffer_get_status ?

```
[GvStream::finalize] n_received_packets     = 81133
[GvStream::finalize] n_missing_packets      = 2084
```

That is a lot of missing packets. How is your camera connected to your machine ?

ChaosCA, 2018-06-08:
Because I can't find a way to access that method given the Python bindings. After I have a [camera object](https://github.com/SintefRaufossManufacturing/python-aravis/blob/master/aravis.py#L19) I attempted to call:

`camera.stream._ar.dll.arv_buffer_get_status()` to no avail..

The camera is ethernet connected

EmmanuelP, 2018-06-08:
> The camera is ethernet connected

Directly, through one or several switches ? A gigabit network ? Using wifi ?

ChaosCA, 2018-06-08:
Ethernet cable, directly.

In this particular example it has a 100Mbit network card.

ChaosCA, 2018-06-08:
After further discussions with JAI they suggested locking the buffer that is being written to as it seems like that could be the issue or increasing the amount of buffers.

I increased the number of buffers to like 30 but then my feed is extremely delayed, and it doesn't seem to push new images at all until you take something off the buffer. Which means I'd get 30 old images before seeing new ones. I thought aravis would remove the oldest buffered image when pushing a new one?

ChaosCA, 2018-06-11:
This seems to be related to this issue: https://github.com/AravisProject/aravis/issues/133

However creating a new stream everytime will cause a memory leak from whate I can tell, and stopping then starting the capture each time is expensive on performance when frequently grabbing images.

EmmanuelP, 2018-06-12:
You don't have to (and should not) create/destroy ArvStream object constantly. Just make sure you are done with the buffer before you push it back to the stream input queue.

closed 2018-06-12 by EmmanuelP

EmmanuelP, 2018-06-12:
One buffer == one image
You may push several buffers in the stream input queue, but if you process the image buffer synchronously, only one buffer is necessary. In this case, the image processing is actually blocking the receiving thread. Which means if your image processing takes longer than the time between the last packet of a frame, and the first packet of the next frame, aravis will probably lose some packets. You can mitigate this issue by increasing the socket buffer size.