AravisProject/aravis #174: Race condition

thecaptury, 2018-06-14:
Hi,

I just found a race condition in arvgcpropertynode.c. It goes something like this:
Thread1: char* tmp = _get_value_data(...);
Thread2: updates value_data: value_data gets freed and reassigned.
Thread1: tries to use tmp, which has been freed by Thread2.

I see two solutions for this:
1. g_strdup value_data returned from _get_value_data(). This is fairly straight forward for most users of _get_value_data(). The only problem is that arv_gc_property_node_get_string() directly returns _get_value_data(). So this would introduce a leak unless arv_gc_property_node_get_string() also returns g_strup()ed values. However, we would then have to update all users of arv_gc_property_node_get_string(). This would also make this function inconsistent with the rest of the library.
We could also add a static buffer into arv_gc_property_node_get_string() and return a pointer to that but that would make it completely thread-unsafe.

2. use reference counted GObjects for value_data. This is a somewhat bigger refactor I believe and will run into similar issues because of the way arv_gc_property_node_get_string() uses the result.

What do you think is the best way to fix this?

cheers,
nils

Address sanitizer trace:

> =================================================================
> 
> ==3524==ERROR: AddressSanitizer: heap-use-after-free on address 0x60200000fb70 at pc 0x7efd5f799ab1 bp 0x7efd1846b460 sp 0x7efd1846ac10
> READ of size 1 at 0x60200000fb70 thread T14
>     #0 0x7efd5f799ab0  (/usr/lib/x86_64-linux-gnu/libasan.so.3+0x44ab0)
>     #1 0x7efd4f4c6c38 in g_str_equal (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x39c38)
>     #2 0x7efd4f4c619f in g_hash_table_lookup (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x3919f)
>     #3 0x7efd5f191945 in _get_pvalue_node /home/captury/scribe/external/aravis/src/arvgcpropertynode.c:246
>     #4 0x7efd5f1920f7 in arv_gc_property_node_set_int64 /home/captury/scribe/external/aravis/src/arvgcpropertynode.c:365
>     #5 0x7efd5f18ab3e in arv_gc_command_execute /home/captury/scribe/external/aravis/src/arvgccommand.c:102
>     #6 0x7efd5f17f355 in arv_device_execute_command /home/captury/scribe/external/aravis/src/arvdevice.c:309
>     arv_camera_start_acquisition(camera);
> 
> 0x60200000fb70 is located 0 bytes inside of 8-byte region [0x60200000fb70,0x60200000fb78)
> freed by thread T0 here:
>     #0 0x7efd5f816a10 in free (/usr/lib/x86_64-linux-gnu/libasan.so.3+0xc1a10)
>     #1 0x7efd5f1918d4 in _get_value_data /home/captury/scribe/external/aravis/src/arvgcpropertynode.c:207
> 
> previously allocated by thread T14 here:
>     #0 0x7efd5f817090 in realloc (/usr/lib/x86_64-linux-gnu/libasan.so.3+0xc2090)
>     #1 0x7efd4f4dcec7 in g_realloc (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4fec7)
> ==3524==ABORTING

thecaptury, 2018-06-15:
I've implemented a fix that

1. locks updating value_data with a mutex

2. pushes values that are replaced into a queue. This essentially results in a controlled leak/delayed use-after-free because we cannot know when it is safe to delete the values. They may propagate all the way to the third party client application, which may keep the strings around forever.

It's rather hacky. So I haven't posted a pull request but I'll start using it internally now.

Any better ideas?

nils

EmmanuelP, 2018-06-15:
Aravis objects are not tread safe. It is not allowed to use an ArvCamera instance (and all its internal objects ArvDevice, ArvGc) from multiple threads. But you can have several ArvCamera instances in different threads. I would not try to fix this race condition, there is probably a lot of other ones.

closed 2018-06-15 by thecaptury