AravisProject/aravis #176: unkown source of frame drop

manuelvolk, 2018-06-26:
Hi me again,

I discover a strange behavior which I can not explain at the moment. 
Therefore, you can hopefully help me out. 

[image]

I write a piece of software which should request an image by a service call and return than the requested image. The class of interest is following: https://github.com/Xamla/camera_aravis/blob/xamla_dev/src/genicam.cpp

Now to the image:

In many cases this process works really well. But sometimes (non-deterministic) the software gets
the frame not after a small amount of time (in my case 60ms) but rather it takes 5-7 seconds. 
This could have multiple reasons. To exclude some of them I start to take a look into the network communication with help of wireshark. As you can see in the image the frame which takes so long to be available in the software is already available after 51 ms on the pc.

Therefore my questions are now:

-What could be the reason for the long delay?
-Is there a tool to further discover what aravis does in my scenario?
-Is my implementation correct?

Some Infos to my config:
aravis 5.12
No performance improvements (no jumbo frames, packet socket or high priority thread)
Distribution Ubuntu 16.04

I'm pretty sure that some of the performance options could help but a delay of 6 seconds can not be
caused by a lower thread priority or what do you think about that?

manuelvolk, 2018-06-27:
After some more testing I also discover that all glib threads seems to be blocked in the case of frame delay / time out.

EmmanuelP, 2018-06-30:
A few comments:
- the new-buffer signal handler is executed in the stream thread, which means any delay here may lead to frame loss/delay, and you should be careful with adat structure concurrent access.
- did you check the status of the returned buffers? Is there missing packets ?

manuelvolk, 2018-06-30:
Hi Emmanuel,

I measure the processing time in my processNewBuffer method and it is takes for every frame only between 300-500 microseconds. Due to the fact that I trigger only single images and between each image request is at least 50ms this is not the problem I guess. Concurrent access could be a problem but the effect I tried to describe already occurs if I only trigger single images from a single camera.
Therefore, no real concurrent access takes place. 

I run the software on the system I describe but in one case on the normal Ubuntu and in the other case in a docker container. Under the normal ubuntu the effect also occurs but here the problem occurs rarely (only every 10000th call). In the docker image the effect occurs much more often (every 50th call).

manuelvolk, 2018-07-19:
Hi EmmanuelP,

after some more testing I guess the source of my problem is just the thread priority 
of the stream thread. I receive all packages as requested an with minimal delays but
some times their are processed in software to late and the frame is already marked as
time out buffer. So one solution could be to increase the priority of the thread the other
to the time out threshold.

The problem is I do not know how to increase the time out threshold. Could you explain
how the buffer time out threshold is defined and how to change it.

EmmanuelP, 2018-07-20:
Hi Manuel,

Have a look at tests/arvcameratest.c and how packet-timeout, realtime and high-priority options are handled.

manuelvolk, 2018-07-23:
Hi Emmanuel,
thank you for your answer. My problem is currently that it is not possible to increase the thread priority with the implemented functions of Aravis, when I run my application in a docker ubuntu image. If someone has already an idea how to do this please share your knowledge with me/us. 

But the other thing is that I still not get when a buffer is marked as ARV_BUFFER_STATUS_TIMEOUT?
https://github.com/AravisProject/aravis/blob/9130ad51e21baf5522100460ec35da00bb4ac3ef/src/arvbuffer.h#L53

How is the timeout defined?

EmmanuelP, 2018-07-23:
Timeout is reached when frame retention times out:

https://github.com/AravisProject/aravis/blob/f471b485281cda23ca9d8930ffe6ce3d865f1e0f/src/arvgvstream.c#L587

You can set the frame retention value using something like:

https://github.com/AravisProject/aravis/blob/f471b485281cda23ca9d8930ffe6ce3d865f1e0f/tests/arvcameratest.c#L357

EmmanuelP, 2018-08-10:
You can also try to increase the socket buffer size:

https://github.com/AravisProject/aravis/blob/ARAVIS_0_5_13/tests/arvcameratest.c#L344-L347

closed 2018-08-10 by EmmanuelP