AravisProject/aravis #177: Checking whether a camera has been acquired

bemeurer, 2018-06-26:
Is there a way in Aravis currently for me to check whether or not a given an `ArvCamera` it has already been acquired by someone else? I'm just trying to get some sort of safeguard when calling `sys::arv_camera_start_acquisition()`

EmmanuelP, 2018-06-30:
Not really. You may try to use arv_device_get_status after start_acquisition in order to check if the command was successful. I did not test this suggestion, please reopen this issue if it doesn't work.

bemeurer, 2018-09-18:
@EmmanuelP I tested this and I don't think it works. Even though the camera fails to acquire I still get the SUCCESS status.

crabsmaps, 2019-05-13:
USB cameras are also affected by this everlasting SUCCESS status.

My guess is that following line prevents setting status of the camera to anything but SUCCESS
https://github.com/AravisProject/aravis/blob/3499bb7ca7f32c06d128ed4b2e1b4e564ceed1ed/src/arvdevice.c#L277

I don't quite understand intentions of this if clause (but I think it would be reasonable to remove it)

EmmanuelP, 2019-05-13:
Device status reports if an error happened during the last actions. This status is reset to SUCCESS when the user call arv_device_get_status. That is why _set_status forbids to set the status to SUCCESS, as it would hide a previous error.

crabsmaps, 2019-05-13:
> That is why _set_status forbids to set the status to SUCCESS

Then it should be checked not like this `device->priv->status == ARV_DEVICE_STATUS_SUCCESS` but rather like this `status == ARV_DEVICE_STATUS_SUCCESS`

EmmanuelP, 2019-05-13:
Oh yes, you are right. I don't use this API myself, as I prefer to check an operation by reading the feature value back. That needs to be fixed.

commit 71d09354 (2019-05-13): device: fix status always set to ARV_DEVICE_STATUS_SUCCESS (#246)

This commit should resolve issues with arv_device_get_status almost always returning ARV_DEVICE_STATUS_SUCCESS, and probably closes #177 .
closed 2019-05-13 by EmmanuelP