AravisProject/aravis #180: aravissrc doesn't return ArvCamera

kartur, 2018-07-20:
Hi,

I use the gstreamer-1.0 plugin of aravis 0.5.13 together with a Xenics thermal camera. 
I got it working with the gstreamer library in C and it works great.

But it would be cool if I could change the camera configuration like I can with the arv-tool for example.

with gst-inspect I found out that aravissrc provides an element "camera":
```
  camera              : Camera instance to retrieve additional information
                        flags: readable
                        Object of type "ArvCamera"
```
But it seems I can't retrieve it with g_object_get. Example: 
`
[...]
data.source_thermal = gst_element_factory_make ("aravissrc", "source_thermal");

g_object_get(data.source_thermal, "name", &thermalCameraName, NULL);
g_print ("thermal name is %s\n", thermalCameraName);

g_object_get(data.source_thermal, "camera", &thermalCamera, NULL);
g_print ("thermal deviceId is %s\n", arv_camera_get_device_id(thermalCamera));
[...]
`

prints:

`thermal name is source_thermal

(gst-test:6109): Aravis-CRITICAL **: 15:02:05.977: arv_camera_get_device_id: assertion 'ARV_IS_CAMERA (camera)' failed
thermal deviceId is (null)`

Is this not the right way to retrieve the ArvCamera object?

EmmanuelP, 2018-07-21:
Changing the MTU size whilethe camera is streaming is not not very nice. I guess we could add a packet size parameter to the aravissrc element. I'm keeping this bug open for this improvement.

Thanks for the report.

EmmanuelP, 2019-08-20:
Implemented in 846d5c22917ab9b1fd51bf2075ca160a507c9233

closed 2019-08-20 by EmmanuelP