AravisProject/aravis #181: Undetected gige camera in 5.13

jewnicorn27, 2018-07-24:
I am using a JAI-080-GE camera, this has two RJ45 connectors and is expected to appear as two distinct cameras. `arv-viewer` only detects one of the two cameras when called while, `arv-viewer -d all:3` gives:

```
Found 0 USB3Vision device (among 4 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.20.222
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = JAI Ltd., Japan
name         = 
model        = AD-080GE_#0
address      = 192.168.20.12
00000000 00 00 00 03 00 f8 ff ff 00 01 00 01 80 00 00 01  ................
00000010 00 00 00 0c df 04 97 08 00 00 00 07 00 00 00 06  ................
00000020 00 00 00 00 00 00 00 00 00 00 00 00 c0 a8 14 0c  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050 4a 41 49 20 4c 74 64 2e 2c 20 4a 61 70 61 6e 00  JAI Ltd., Japan.
00000060 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000070 41 44 2d 30 38 30 47 45 5f 23 30 00 00 00 00 00  AD-080GE_#0.....
00000080 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090 30 2e 33 2e 31 2e 37 00 00 00 00 00 00 00 00 00  0.3.1.7.........
000000a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000b0 44 69 67 69 74 61 6c 20 32 43 43 44 20 50 72 6f  Digital 2CCD Pro
000000c0 67 72 65 73 73 69 76 65 20 53 63 61 6e 00 00 00  gressive Scan...
000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000e0 58 38 30 31 38 37 30 00 00 00 00 00 00 00 00 00  X801870.........
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device 'JAI Ltd., Japan-X801870' found (interface 192.168.20.222) user_name '' - MAC_name 'JAI Ltd., Japan-X801870'
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = JAI Ltd., Japan
name         = 
model        = AD-080GE_#1
address      = 192.168.20.7
00000000 00 00 00 03 00 f8 ff ff 00 01 00 01 80 00 00 01  ................
00000010 00 00 00 0c df 04 a7 08 00 00 00 07 00 00 00 06  ................
00000020 00 00 00 00 00 00 00 00 00 00 00 00 c0 a8 14 07  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050 4a 41 49 20 4c 74 64 2e 2c 20 4a 61 70 61 6e 00  JAI Ltd., Japan.
00000060 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000070 41 44 2d 30 38 30 47 45 5f 23 31 00 00 00 00 00  AD-080GE_#1.....
00000080 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090 30 2e 33 2e 31 2e 37 00 00 00 00 00 00 00 00 00  0.3.1.7.........
000000a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000b0 44 69 67 69 74 61 6c 20 32 43 43 44 20 50 72 6f  Digital 2CCD Pro
000000c0 67 72 65 73 73 69 76 65 20 53 63 61 6e 00 00 00  gressive Scan...
000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000e0 58 38 30 31 38 37 30 00 00 00 00 00 00 00 00 00  X801870.........
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device 'JAI Ltd., Japan-X801870' found (interface 192.168.20.222) user_name '' - MAC_name 'JAI Ltd., Japan-X801870'
```
Both cameras appear to have the same MAC_name, however have different static IPs, is there a way to interface with them?

EmmanuelP, 2018-07-25:
I did not foresee this case. I working on a patch that will use vendor-model-serial as the main device hash table key.

WIP here: https://github.com/AravisProject/aravis/tree/bugs/181

jewnicorn27, 2018-07-25:
You are a wonderful person! Thank you for this fix. I've tested it, and seems to do exactly what I want.
[image]

EmmanuelP, 2018-07-26:
Fixed in master. Thanks for the report.

closed 2018-07-26 by EmmanuelP