AravisProject/aravis #182: "num-buffers" does not work with aravissrc

bwkwok, 2018-07-25:
I'm testing a usb3 camera and using the gstreamer plugin to save videos.  However, even specifying a "num-buffers" value for the aravis src, the capturing kept going unless I stopped it by ctrl-c.

`gst-launch-1.0 --gst-plugin-path=/usr/local/lib/ aravissrc camera-name="PointGrey-00DC1E7A" num=buffers=300 ! capsfilter caps="video/x-raw,format=RGB,framerate=30/1" ! videoconvert ! capsfilter caps="video/x-raw,format=I420" ! avenc_mpeg4 ! mp4mux ! filesink location=~/test.mp4`

I expected the above should stop after acquiring 300 buffers (10 sec) at a framerate of 30fps.

OTOH, it works with videotestsrc:

`gst-launch-1.0 videotestsrc num=buffers=300 ! capsfilter caps="video/x-raw,format=RGB,framerate=30/1" ! videoconvert ! capsfilter caps="video/x-raw,format=I420" ! avenc_mpeg4 ! mp4mux ! filesink location=~/test.mp4`

bwkwok, 2018-07-31:
Hi @EmmanuelP. THank you for the comment.

I take it as the "num-buffers" element for aravissrc is not supported at this moment?

bwkwok, 2018-08-03:
In the meantime, is there a similar element/config for "aravissrc" to acquire upto a certain number of frames or number of seconds?

EmmanuelP, 2020-08-13:
Fixed by 8857d907aa40dab9d920f6ae03891d7f18d1a68b

closed 2020-08-13 by EmmanuelP