AravisProject/aravis #183: Read/write issue with USB3Vision Smartek UCC4113C

EmmanuelP, 2018-07-31:
Here is what Patrick reported to the mailing list:

> My name is Patrick Price, I’m working on integrating a USB3 Vision camera with a Qualcomm 820 running Debian Linux. I’m trying to stream from a Smartek UCC4113C using Aravis 0.6, but I’m running into some trouble. I am able to connect to the camera and query it’s features but when I try to stream from it with arv-viewer I get the error “GStreamer cannot understand the camera pixel format: 0x0!” When I query the features it comes back with a list of pixel formats but all of them marked “(Not Available)”. The errors match pretty closely to this thread: https://www.freelists.org/post/aravis/gstreamer-format-was-aravisviewer-SIGSEGV, which suggests there’s something wrong with the GenICam file. I’ve played with some of the example code on your wiki but run into similar issues. I’m new to GenICam and USB3 Vision, so I’m at a loss.

EmmanuelP, 2018-07-31:
It looks like there is an issue while reading/writing registers, as seen in aravis-viewer.txt:
```
[[UvDevice::write_memory] Unexpected answer (0x8003)
[[UvDevice::write_memory] Unexpected answer (0x8003)
[[UvDevice::write_memory] Unexpected answer (0x8003)
[[UvDevice::write_memory] Unexpected answer (0x8003)
[[UvDevice::write_memory] Unexpected answer (0x8003)
[[UvDevice::read_memory] Unexpected answer (0x8003)
```
That probably explains the fact all pixel formats are marked as unavailable.

discwv, 2018-08-01:
Poking around in the registers, maybe something to do with endianess? If I read through the Genicam file and check the relevant registers by hand for the pixel format, I find the correct values.

There are registers that contain values pointing to other registers. When I look at the value in one using it's address, I get the correct address of the register it's pointing at. If I use the name, the endianess is reversed.

```
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control CRegsProviderBaseAddresses_BaseAddress
Smartek Vision-10000045 (USB3)
CRegsProviderBaseAddresses_BaseAddress = 61440
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control CImagePreprocessingReg
Smartek Vision-10000045 (USB3)
CImagePreprocessingReg = 485621760
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control CPixelFormatBaseReg
Smartek Vision-10000045 (USB3)
CPixelFormatBaseReg = 0
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control PixelFormatReg
Smartek Vision-10000045 (USB3)
PixelFormatReg = 0
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control R[0xf014]
Smartek Vision-10000045 (USB3)
R[0x0000f014] = 0x0000f21c
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control R[0xf22c]
Smartek Vision-10000045 (USB3)
R[0x0000f22c] = 0x0000f2d8
linaro@linaro-alip:~$ arv-tool-0.6 --name="Smartek Vision-10000045" control R[0xf2dc]
Smartek Vision-10000045 (USB3)
R[0x0000f2dc] = 0x00000444
```
Notice "R[0x0000f014] = 0x0000f21c" but "CImagePreprocessingReg = 485621760" = 0x1cf20000

The relevant sections from the genicam file:
```
<Integer Name="CRegsProviderBaseAddresses_BaseAddress" NameSpace="Custom">
	<ToolTip>Returns the base address of this register provider</ToolTip>
	<Description>Returns the base address of this register provider</Description>
	<DisplayName>CRegsProviderBaseAddresses_BaseAddress</DisplayName>
	<Visibility>Invisible</Visibility>
	<Value>0xf000</Value>
</Integer>

<IntReg Name="CImagePreprocessingReg">
	<Description>CImagePreprocessingReg</Description>
	<Visibility>Invisible</Visibility>
	<pAddress>CRegsProviderBaseAddresses_BaseAddress</pAddress>
	<Address>0x14</Address>
	<Length>4</Length>
	<AccessMode>RO</AccessMode>
	<pPort>Device</pPort>
	<Sign>Unsigned</Sign>
	<Endianess>LittleEndian</Endianess>
</IntReg>

<IntReg Name="CPixelFormatBaseReg">
	<Description>CPixelFormatBaseReg</Description>
	<Visibility>Invisible</Visibility>
	<pAddress>CImagePreprocessingReg</pAddress>
	<Address>0x10</Address>
	<Length>4</Length>
	<AccessMode>RO</AccessMode>
	<pPort>Device</pPort>
	<Sign>Unsigned</Sign>
	<Endianess>LittleEndian</Endianess>
</IntReg>
	
<IntReg Name="PixelFormatReg">
	<Description>PixelFormatReg</Description>
	<Visibility>Invisible</Visibility>
	<pAddress>CPixelFormatBaseReg</pAddress>
	<Address>0x0</Address>
	<Length>4</Length>
	<AccessMode>RW</AccessMode>
	<pPort>Device</pPort>
	<pInvalidator>UserSetLoad</pInvalidator>
	<pInvalidator>ReverseX</pInvalidator>
	<pInvalidator>ReverseY</pInvalidator>
	<Sign>Unsigned</Sign>
	<Endianess>LittleEndian</Endianess>
</IntReg>
```

Any thoughts? Somewhere I should look?

EmmanuelP, 2018-08-02:
I think the problem is in this code:

https://github.com/AravisProject/aravis/blob/cc62161becc317e5143deb862652983466302dae/src/arvgcport.c#L106-L118

Coud you attach the content of the RegisterDescription node at the start of the genicam data of your camera, using `arv-tool-0.6 genicam` ?

EmmanuelP, 2018-08-02:
Could you also check if the genicam data has a DeviceEndianessMechanism node ?

Meanwhile, you could try to make _register_workaround always return FALSE and see if it fixes your issue. Obviously it is not the right fix, but it will help us understand if we are looking at the right place.

discwv, 2018-08-02:
There was no DeviceEndianessMechanism node. The RegisterDescription node:
```
<RegisterDescription
	ModelName="twentynine"
	VendorName="Smartek"
	ToolTip="Smartek U3V Vision camera interface"
	StandardNameSpace="None"
	SchemaMajorVersion="1"
	SchemaMinorVersion="0"
	SchemaSubMinorVersion="1"
	MajorVersion="3"
	MinorVersion="0"
	SubMinorVersion="0"
	ProductGuid="E35AF211-7EF3-4dfe-AE55-011298CF0A0E"
	VersionGuid="5A9959AD-F3DD-46bb-8C9D-2DCABB5C635B"
	xmlns="http://www.genicam.org/GenApi/Version_1_0"
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xsi:schemaLocation="http://www.genicam.org/GenApi/Version_1_0 GenApiSchema.xsd">
```
Looks like the schema version is less than 1.1.0, and all these registers are length 4. I'll try that _register_workaround change.

discwv, 2018-08-02:
I made the following change and was able to stream from the camera.
```
static gboolean
_register_workaround_check (ArvGcPort *port, guint64 length)
{
	return FALSE;
	//ArvDomDocument *document;
	//ArvGcRegisterDescriptionNode *register_description;

	//document = arv_dom_node_get_owner_document (ARV_DOM_NODE (port));
	//register_description = ARV_GC_REGISTER_DESCRIPTION_NODE (arv_dom_document_get_document_element (document));

	//return length == 4 && (arv_gc_register_description_node_compare_schema_version (register_description, 1, 1, 0) < 0);
}
```

This check shouldn't exist for USB3 Vision cameras, right? Looks like the standard only mentions this for GigE Vision.

EmmanuelP, 2018-08-02:
> Looks like the standard only mentions this for GigE Vision.

Oh, you are right. It is written everywhere in appendix 3, but it did not reach my brain. I will write a fix tomorrow. Thanks.

EmmanuelP, 2018-08-03:
This should be fixed by 994abcd8910550ec816b321c4cb53039e52b6591 .

Thanks for the report. Cheers.

closed 2018-08-03 by EmmanuelP