AravisProject/aravis #184: Scheduled Action Commands

thecaptury, 2018-08-09:
GigEVision 2.0 introduces Action Commands that will be processed by the camera at a specified time.

From Basler's documentation it looks like they just add a timestamp field to the command packet.

https://www.baslerweb.com/en/sales-support/downloads/document-downloads/multi-camera-applications/

That can't be too hard to figure out. It'll be a bit of work to do it properly though.

EmmanuelP, 2019-08-15:
Hi Nils,

I have seen you worked on this feature. Is it in a working state ?

shakthisharavanan, 2021-01-21:
Any update on aravis support for this? I have some lucid cameras that also support this.
https://support.thinklucid.com/app-note-multi-camera-synchronization-using-ptp-and-scheduled-action-commands/

mbokeloh-i, 2022-12-03:
Hi Nils :-)

I'm also curious whether it's possible now to use this feature with Aravis.

ikazinov, 2023-12-16:
Hi all,
Any update on aravis support for this feature ??