AravisProject/aravis #188: Does the viewer require gstreamer 1.0?

rehno-lindeque, 2018-08-15:
Build snippets using `--enable-viewer --enable-gst-0.10-plugin`:

```
  Build USB support:            yes
  Build packet socket support:  yes
  Build viewer:                 no
  Build gstreamer plugin:       no
  Build gstreamer-0.10 plugin:  yes
  Build with fast heartbeat:    no
  Build with address sanatizer: no
```
...

```
checking for gstreamer-base-1.0 gstreamer-app-1.0 gstreamer-video-1.0 gtk+-3.0 >= 3.12 libnotify... no
configure: error: No package 'gstreamer-base-1.0' found
No package 'gstreamer-app-1.0' found
```

I had expected to see it use `gstreamer-app-0.10` instead of `gstreamer-app-1.0`.

EmmanuelP, 2018-08-15:
Hi,

Aravis viewer uses gstreamer 1.0.

The `gstreamer-0.10 plugin` message is only about the aravis gstreamer plugin, which is not what is used by the viewer.

Cheers.

closed 2018-08-15 by EmmanuelP

rehno-lindeque, 2018-08-17:
Thank you. I've created a package for the nix package manager: https://github.com/NixOS/nixpkgs/pull/45274