AravisProject/aravis #191: arv_gc_string_get_value fails for current enum entry

TIS-Edgar, 2018-08-30:
Hi Emmanuel,

I am unable to retrieve the current value for an enumeration for a 'The Imaging Source DFK Z12GX236'. You can find the genicam description [here](https://gist.github.com/TIS-Edgar/9db85e6a403b6acd9619823402bbe466).

On current master this is the observed behavior:

    > src/arv-tool-0.6 control StrobeOperation -n TIS-15510006

    TIS-15510006 (192.168.0.149)
    StrobeOperation = (null)

With git bisect I was able to identify commit 25dbaa6 as the one causing the bug.
After checking out the previous version ( cf32efb ) the output is correct:

    TIS-15510006 (192.168.0.149)
    StrobeOperation = Exposure

Since I am unsure on how to fix this, I hope you can advise.

Regards,

Edgar

EmmanuelP, 2018-08-30:
Hi Edgar,

Could you please attach the output of `arv-tool-0.6 control StrobeOperation -n TIS-15510006 --debug genicam:3` ?

TIS-Edgar, 2018-08-31:
Hi Emmanuel,

Here you are:

```[Gc::register_feature_node] Register node 'Device' [Port]
[Gc::register_feature_node] Register node 'Root' [Category]
[Gc::register_feature_node] Register node 'AcquisitionControl' [Category]
[Gc::register_feature_node] Register node 'AcquisitionStart' [Command]
[Gc::register_feature_node] Register node 'AcquisitionStop' [Command]
[Gc::register_feature_node] Register node 'AcquisitionMode' [Enumeration]
[Gc::register_feature_node] Register node 'AcquisitionStartStopRegister' [IntReg]
[Gc::register_feature_node] Register node 'TransportLayer' [Category]
[Gc::register_feature_node] Register node 'CalcBytesPerPixel' [IntSwissKnife]
[Gc::register_feature_node] Register node 'CalcPayloadSize' [IntSwissKnife]
[Gc::register_feature_node] Register node 'PayloadSize' [Integer]
[Gc::register_feature_node] Register node 'TLParamsLocked' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSPacketSize' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSFireTestPacket' [Command]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragment' [Boolean]
[Gc::register_feature_node] Register node 'GevTimestampTickFrequency' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSPacketSizeRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSFireTestPacketRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragmentRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevTimestampTickFrequencyLowRegister' [IntReg]
[Gc::register_feature_node] Register node 'DeviceControl' [Category]
[Gc::register_feature_node] Register node 'TimestampReset' [Command]
[Gc::register_feature_node] Register node 'TimestampLatch' [Command]
[Gc::register_feature_node] Register node 'TimestampLatchValue' [Integer]
[Gc::register_feature_node] Register node 'TimestampLatchValueBuilder' [IntSwissKnife]
[Gc::register_feature_node] Register node 'TimestampControlRegister' [IntReg]
[Gc::register_feature_node] Register node 'TimestampLatchedHi' [IntReg]
[Gc::register_feature_node] Register node 'TimestampLatchedLo' [IntReg]
[Gc::register_feature_node] Register node 'DriverPropertySets' [Integer]
[Gc::register_feature_node] Register node 'DriverDataAddress' [Integer]
[Gc::register_feature_node] Register node 'DriverDataLength' [Integer]
[Gc::register_feature_node] Register node 'ImageFormat' [Category]
[Gc::register_feature_node] Register node 'WDR' [Boolean]
[Gc::register_feature_node] Register node 'SensorWidth' [Integer]
[Gc::register_feature_node] Register node 'SensorHeight' [Integer]
[Gc::register_feature_node] Register node 'WidthMax' [IntSwissKnife]
[Gc::register_feature_node] Register node 'HeightMax' [IntSwissKnife]
[Gc::register_feature_node] Register node 'WidthMaxValue' [IntSwissKnife]
[Gc::register_feature_node] Register node 'HeightMaxValue' [IntSwissKnife]
[Gc::register_feature_node] Register node 'Width' [Integer]
[Gc::register_feature_node] Register node 'Height' [Integer]
[Gc::register_feature_node] Register node 'OffsetX' [Integer]
[Gc::register_feature_node] Register node 'OffsetY' [Integer]
[Gc::register_feature_node] Register node 'ManualOffsetLocked' [IntSwissKnife]
[Gc::register_feature_node] Register node 'OffsetAutoCenter' [Enumeration]
[Gc::register_feature_node] Register node 'BinningValue' [Integer]
[Gc::register_feature_node] Register node 'BinningHorizontal' [Integer]
[Gc::register_feature_node] Register node 'BinningVertical' [Integer]
[Gc::register_feature_node] Register node 'MaxDataRate' [Float]
[Gc::register_feature_node] Register node 'CurrentPixelFormatBits' [SwissKnife]
[Gc::register_feature_node] Register node 'CurrentFormatBytes' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPSFromDataRate' [SwissKnife]
[Gc::register_feature_node] Register node 'UnboundedHmaxMin' [SwissKnife]
[Gc::register_feature_node] Register node 'HmaxMin' [SwissKnife]
[Gc::register_feature_node] Register node 'Vmax' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPSFromSensorTiming' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPS' [SwissKnife]
[Gc::register_feature_node] Register node 'MinFPS' [SwissKnife]
[Gc::register_feature_node] Register node 'PixelClock' [Integer]
[Gc::register_feature_node] Register node 'FPSToFP1KS' [Converter]
[Gc::register_feature_node] Register node 'FPS' [Float]
[Gc::register_feature_node] Register node 'TestBinningValue' [Integer]
[Gc::register_feature_node] Register node 'TestBinningHorizontal' [Integer]
[Gc::register_feature_node] Register node 'TestBinningVertical' [Integer]
[Gc::register_feature_node] Register node 'TestWidth' [Integer]
[Gc::register_feature_node] Register node 'TestHeight' [Integer]
[... 136 lines cut ...]
[Gc::register_feature_node] Register node 'DeviceVendorName' [StringReg]
[Gc::register_feature_node] Register node 'DeviceModelName' [StringReg]
[Gc::register_feature_node] Register node 'DeviceVersion' [StringReg]
[Gc::register_feature_node] Register node 'DeviceManufacturerInfo' [StringReg]
[Gc::register_feature_node] Register node 'DeviceID' [StringReg]
[Gc::register_feature_node] Register node 'NumberOfStreamChannels' [IntReg]
[Gc::register_feature_node] Register node 'GevSCPHostPort' [Integer]
[Gc::register_feature_node] Register node 'GevSCPHostPortReg' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragmentReg' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSBigEndian' [Boolean]
[... cut ...]
[Gc::register_feature_node] Register node 'GevSCPAddrCalc' [IntSwissKnife]
[Gc::register_feature_node] Register node 'GevStreamChannelSelector' [Integer]
[Gc::register_feature_node] Register node 'GevStreamChannelSelectorMax' [IntSwissKnife]
[GcRegisterNode::_get_integer_value] reglsb = 29, regmsb, 29, lsb = 2, msb = 2
[GcRegisterNode::_get_integer_value] value = 0x00000004
[GcRegisterNode::_get_integer_value] mask  = 0x00000004
[GcRegisterNode::_get_integer_value] address = 0xff000120, value = 0xffffffffffffffff
StrobeOperation = (null)
```

EmmanuelP, 2018-08-31:
Could you also attach the debug output in the working case ?

TIS-Edgar, 2018-08-31:
Here the working output with commit cf32efb:
```
TIS-15510006 (192.168.0.149)
[Gc::register_feature_node] Register node 'Device' [Port]
[Gc::register_feature_node] Register node 'Root' [Category]
[Gc::register_feature_node] Register node 'AcquisitionControl' [Category]
[Gc::register_feature_node] Register node 'AcquisitionStart' [Command]
[Gc::register_feature_node] Register node 'AcquisitionStop' [Command]
[Gc::register_feature_node] Register node 'AcquisitionMode' [Enumeration]
[Gc::register_feature_node] Register node 'AcquisitionStartStopRegister' [IntReg]
[Gc::register_feature_node] Register node 'TransportLayer' [Category]
[Gc::register_feature_node] Register node 'CalcBytesPerPixel' [IntSwissKnife]
[Gc::register_feature_node] Register node 'CalcPayloadSize' [IntSwissKnife]
[Gc::register_feature_node] Register node 'PayloadSize' [Integer]
[Gc::register_feature_node] Register node 'TLParamsLocked' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSPacketSize' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSFireTestPacket' [Command]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragment' [Boolean]
[Gc::register_feature_node] Register node 'GevTimestampTickFrequency' [Integer]
[Gc::register_feature_node] Register node 'GevSCPSPacketSizeRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSFireTestPacketRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragmentRegister' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevTimestampTickFrequencyLowRegister' [IntReg]
[Gc::register_feature_node] Register node 'DeviceControl' [Category]
[Gc::register_feature_node] Register node 'TimestampReset' [Command]
[Gc::register_feature_node] Register node 'TimestampLatch' [Command]
[Gc::register_feature_node] Register node 'TimestampLatchValue' [Integer]
[Gc::register_feature_node] Register node 'TimestampLatchValueBuilder' [IntSwissKnife]
[Gc::register_feature_node] Register node 'TimestampControlRegister' [IntReg]
[Gc::register_feature_node] Register node 'TimestampLatchedHi' [IntReg]
[Gc::register_feature_node] Register node 'TimestampLatchedLo' [IntReg]
[Gc::register_feature_node] Register node 'DriverPropertySets' [Integer]
[Gc::register_feature_node] Register node 'DriverDataAddress' [Integer]
[Gc::register_feature_node] Register node 'DriverDataLength' [Integer]
[Gc::register_feature_node] Register node 'ImageFormat' [Category]
[Gc::register_feature_node] Register node 'WDR' [Boolean]
[Gc::register_feature_node] Register node 'SensorWidth' [Integer]
[Gc::register_feature_node] Register node 'SensorHeight' [Integer]
[Gc::register_feature_node] Register node 'WidthMax' [IntSwissKnife]
[Gc::register_feature_node] Register node 'HeightMax' [IntSwissKnife]
[Gc::register_feature_node] Register node 'WidthMaxValue' [IntSwissKnife]
[Gc::register_feature_node] Register node 'HeightMaxValue' [IntSwissKnife]
[Gc::register_feature_node] Register node 'Width' [Integer]
[Gc::register_feature_node] Register node 'Height' [Integer]
[Gc::register_feature_node] Register node 'OffsetX' [Integer]
[Gc::register_feature_node] Register node 'OffsetY' [Integer]
[Gc::register_feature_node] Register node 'ManualOffsetLocked' [IntSwissKnife]
[Gc::register_feature_node] Register node 'OffsetAutoCenter' [Enumeration]
[Gc::register_feature_node] Register node 'BinningValue' [Integer]
[Gc::register_feature_node] Register node 'BinningHorizontal' [Integer]
[Gc::register_feature_node] Register node 'BinningVertical' [Integer]
[Gc::register_feature_node] Register node 'MaxDataRate' [Float]
[Gc::register_feature_node] Register node 'CurrentPixelFormatBits' [SwissKnife]
[Gc::register_feature_node] Register node 'CurrentFormatBytes' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPSFromDataRate' [SwissKnife]
[Gc::register_feature_node] Register node 'UnboundedHmaxMin' [SwissKnife]
[Gc::register_feature_node] Register node 'HmaxMin' [SwissKnife]
[Gc::register_feature_node] Register node 'Vmax' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPSFromSensorTiming' [SwissKnife]
[Gc::register_feature_node] Register node 'MaxFPS' [SwissKnife]
[Gc::register_feature_node] Register node 'MinFPS' [SwissKnife]
[Gc::register_feature_node] Register node 'PixelClock' [Integer]
[Gc::register_feature_node] Register node 'FPSToFP1KS' [Converter]
[Gc::register_feature_node] Register node 'FPS' [Float]
[Gc::register_feature_node] Register node 'TestBinningValue' [Integer]
[Gc::register_feature_node] Register node 'TestBinningHorizontal' [Integer]
[Gc::register_feature_node] Register node 'TestBinningVertical' [Integer]
[Gc::register_feature_node] Register node 'TestWidth' [Integer]
[Gc::register_feature_node] Register node 'TestHeight' [Integer]
[Gc::register_feature_node] Register node 'TestPixelFormat' [Integer]
[... 136 lines cut ...]
[Gc::register_feature_node] Register node 'DeviceModelName' [StringReg]
[Gc::register_feature_node] Register node 'DeviceVersion' [StringReg]
[Gc::register_feature_node] Register node 'DeviceManufacturerInfo' [StringReg]
[Gc::register_feature_node] Register node 'DeviceID' [StringReg]
[Gc::register_feature_node] Register node 'NumberOfStreamChannels' [IntReg]
[Gc::register_feature_node] Register node 'GevSCPHostPort' [Integer]
[Gc::register_feature_node] Register node 'GevSCPHostPortReg' [MaskedIntReg]
[Gc::register_feature_node] Register node 'GevSCPSDoNotFragmentReg' [MaskedIntReg]
[... cut ...]
[Gc::register_feature_node] Register node 'GevStreamChannelSelector' [Integer]
[Gc::register_feature_node] Register node 'GevStreamChannelSelectorMax' [IntSwissKnife]
[GcRegisterNode::_get_integer_value] reglsb = 29, regmsb, 29, lsb = 2, msb = 2
[GcRegisterNode::_get_integer_value] value = 0x00000004
[GcRegisterNode::_get_integer_value] mask  = 0x00000004
[GcRegisterNode::_get_integer_value] address = 0xff000120, value = 0x1
[GcEnumeration::get_string_value] value = 1 - string = Exposure
StrobeOperation = Exposure
```

EmmanuelP, 2018-08-31:
And could you also attach the genicam data, if it is possible, or at least the relevant part.
I think the problem lies in the default sign of IntReg, MaskedIntReg and StructEntry nodes, which should be unsigned by default.

TIS-Edgar, 2018-08-31:
Do you mean [the genicam description](https://gist.github.com/TIS-Edgar/9db85e6a403b6acd9619823402bbe466) from my initial post?

EmmanuelP, 2018-08-31:
Oh yes, sorry.

EmmanuelP, 2018-09-03:
Hi Edgar,

Could you please test this branch: https://github.com/AravisProject/aravis/tree/wip/default-signedness

TIS-Edgar, 2018-09-03:
Hi Emmanuel,

I can confirm that the new branch fixes the issue with StrobeOperation.

EmmanuelP, 2018-09-03:
Pushed to master. Thanks.

closed 2018-09-03 by EmmanuelP