AravisProject/aravis #192: arv_stream_free_buffers

werner-Z, 2018-09-04:
API function (for GigE cams) to free all the buffers, not only the one in the input and output queues, but also the one handled by the packet receiving htread.
Reason: In our current application we have to handle frequent changes of the read-out-region for a grab in a grab-preparation phase some time before the effective image aquisition starts. Some times it can happen that an already prepared snap is aborted before capture and in this case the buffers, which never will be grabbed in, have to be removed.
I already prepared a implementation in a branch, which could be used for a pull request.

EmmanuelP, 2018-09-11:
See PR https://github.com/AravisProject/aravis/pull/193

EmmanuelP, 2019-08-18:
It was fixed when thread start/stop was merged into master.

closed 2019-08-18 by EmmanuelP