AravisProject/aravis #199: auto packet size does not consider possible value changes by device

TIS-Edgar, 2018-10-11:
Hi Emmanuel,

We had some problems with `arv_gv_device_auto_packet_size`.
We noticed that aravis regularly determined a packet size that is far under the allowed maximum.
For example with a MTU of 9000 aravis would still determine a packet size of 3708 bytes although jumbo frames go through everywhere.

From our understanding the function needs to read the GevSCPSPacketSize back after setting it as the device is allowed to round the value as it has to be divisible by 4.

```
arv_device_set_integer_feature_value (device, "GevSCPSPacketSize", current_size);
current_size = arv_device_get_integer_feature_value(device, "GevSCPSPacketSize");
```
This seems to fix our issues, but I am unsure if there is something I am overlooking or if you want additional checks, etc.

EmmanuelP, 2018-10-11:
Hi Edgar,

Your fix seems reasonable. Could you create a pull request ?

TIS-Edgar, 2018-10-11:
Sure, give me a moment.

closed 2018-10-11 by EmmanuelP