AravisProject/aravis #201: packet size negotiations fails for some devices

TIS-Edgar, 2018-10-12:
Hi Emmanuel,

There seem to be some problems with the auto negotiation of the packet size.

Older cameras respond differently to wrong packet sizes. I have been made aware of this issue only today or I would have described it before my patch, so sorry for that.

You can find a log of an attempt to stream from such a camera here (started with ARV_DEBUG=device:3 aborted with ctrl-c ): https://gist.github.com/TIS-Edgar/eb4c062e36a24dd67d76b17bac93bf12

The answer 0x8005 you see there is according to wireshark a GEV_STATUS_BAD_ALIGNMENT.

I tried using arv_device_get_status but it always returns 0.

I assume this is due to
```
if (device->priv->status == ARV_DEVICE_STATUS_SUCCESS)
	return;
```
in the _set_state function, but I may be wrong.

Is there another way to retrieve error to implement error handling?

Another way of working around the negotiation problems is to disable 
`current_size = arv_device_get_integer_feature_value(device, "GevSCPSPacketSize");`
and introduce a modulo 4 check at the end of the main loop. This works for TIS cameras as they are always aligned by 4. I am not sure if there are devices which behave different.

How can `arv_gv_device_auto_packet_size` be adjusted to remain at the upper end of the possible packet size range?
How can error handling be implemented in a way that allows reacting to such errors?

highperformancecoder, 2019-08-15:
I had a bit of a look at the packet size auto negotiation with eBUSPlayer (free windows download, or a paid Linux version) with wireshark. It seems it is writing a series of different packet sizes into the Stream Channel #0 Packet Size register, with a bit set to fire a test packet, and request no fragmentation. See attached image. But Aravis accepts the write register command, but because it doesn't know about that register, the value is ignored, same with the flags, so no test packet is fired. Eventually, eBUSPlayer gives up, and reports setting the packet size to the fallback value of 1476, although curiously in the trace it sets it to 2000 (which causes fragmentation on a standard MTU of 1500).

highperformancecoder, 2019-08-15:
[image]

highperformancecoder, 2019-08-15:
NB the only way I have been able to get aravis fake GV camera to work with eBUSPlayer is to clamp the value of data_size on line 363 of arvgvfakecamera.c to 16000. This value was determined empirically, I have no theory as to why it should be that.  I'm not even sure what the variable gv_packet_size is supposed to represent. Jumbo packets would be fine with that value, but the maximum sized UDP packet is in theory 2^16, so this limitation reduces the effectiveness of loopback device connection (which eBUSPlayer refuses to recognise anyway).

EmmanuelP, 2019-08-15:
@highperformancecoder your comments are not related to this issue. I have created a new issue about fake camera TestPacket support.

@TIS-Edgar I guess we could only try sizes multiple of 4, or even 8. Even if we know the GevSCPSPacketSize write failed, what could we do ? Or may be the GevSCPSPacketSize feature has an `<Inc>` node we could use ?

EmmanuelP, 2019-08-15:
It looks like we could use the `<Inc>` node value. Looking at the Genicam documentation, it is an indicator of the allowed values. Sometimes a `<ValueSet>` is used, but this one is not implemented in aravis.

EmmanuelP, 2019-08-15:
@TIS-Edgar could you check the camera requiring GevSCPSPacketSize to be a multiple of 4 have an <Inc> node set to 4.

EmmanuelP, 2019-08-29:
Hi @TIS-Edgar , I've just pushed a fix to master that takes `<Inc>` property into account while scanning for working packet sizes.

closed 2019-08-29 by EmmanuelP