AravisProject/aravis #204: arv_gv_interface_camera_locate() fails if docker service is running

thecaptury, 2018-10-16:
Essentially, arv_gv_interface_camera_locate() tries to reach a given camera on all available interfaces and if it receives a reply, it assumes that this is a good interface to reach the camera. Makes sense.

However, docker installs a bridge interface docker0 that essentially forwards everything to eth0. Replies are also forwarded correctly. This is done by iptables. In my case the setup looks like this:

eth0: 10.90.90.42/24
docker0: 172.17.42.1/16
camera: 10.90.90.45

arv_gv_interface_camera_locate() sends a packet to docker0, which gets forwarded to eth0 and from there sent to the camera. The camera replies to 10.90.90.42 because the forwarding rewrote the packet. This gets forwarded back correctly. arv_gv_interface_camera_locate() is happy and uses docker0 as the interface address.

But that interface_address is used to tell the camera where to send 1. test packets for auto packet size and 2. for telling it where to stream the images to. The camera has no clue where to send the packet to because it's in a completely different subnet and doesn't send anything. This also means that running aravis inside a docker image, to talk to a real camera will not work without additional network configuration...

I see several ways to fix this issue:
1. Tell people to properly configure their networking. But I'm sure other people will have docker installed and have no clue what's going on and I found this pretty hard to find. Everything works fine until you try to stream. Then the camera, depending on the model, starts acting weird, disconnects or just ignores the request.

2. blacklist interfaces called docker0 and while we're at it also veth*.

3. In the past I've used a different method to find the network interface that matches a given IP: Iterate over interfaces and then check

if ((device_address & netmask) == (interface_address & netmask)) { /* found it */ }

This is actually a lot more restrictive because it doesn't allow any routing but I don't think one should use routing when dealing with cameras. That just causes unnecessary overhead. It could also fail if multiple network interfaces have overlapping ip ranges.

I can prepare a pull request for 2 or 3.

cheers,
nils

EmmanuelP, 2018-10-17:
Hi Nils,

I would go with the third proposal, with an additional step where if you don't find an interface that fulfills the restriction, you take one of the other that has replied to the discovery packet.

Thanks for your work on this issue.

thecaptury, 2018-10-17:
Hi Emmanuel,

over night I came up with two more options:

4. use the SendTestPacket infrastructure.

5. Provide a way for the user of the library to specify the interface address. My software already knows which interface it is on because we have our own camera discovery.

3 + the existing method as a fallback should work though. I'll prepare a pull request for this. This may take a couple of days though. I'm very busy at the moment.

nils

EmmanuelP, 2018-12-12:
Pushed to master.

Thanks Nils.

closed 2018-12-12 by EmmanuelP