AravisProject/aravis #205: Cannot read XML from LW-AL-CMV4000M

ben2789, 2018-10-23:
I am working with a [LW-AL-CMV4000M](http://isgcameras.com/product/lw-al-cmv4000m-usb3-mono-smart-camera/) from ISG cameras. When I attempt to read the XML from the camera with `arv-test-0.6` I get the following error:

```Testing device 'LightWise Allegro CMV4000 Mono' from 'Imaging Solutions Group'

(process:5082): Aravis-CRITICAL **: 13:43:40.805: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed

(process:5082): Aravis-CRITICAL **: 13:43:40.805: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
Genicam: FAILED Invalid Genicam XML data
Sensor size read: 0 0```

setting the environment variable `ARV_DEBUG=all:3` gives the following output:

[ArvDomNode::append_child] Can't append '#text' to 'RegisterDescription'
packet_type  = cmd
command      = read-memory-cmd
size         = 12
id           = 65335
address      = 0x0000000000000004
size         =         64 (0x00000040)
00000000 55 33 56 43 00 40 00 08 0c 00 37 ff 04 00 00 00  U3VC.@....7.....
00000010 00 00 00 00 00 00 40 00                          ......@.        
packet_type  = ack
command      = read-memory-ack
size         = 64
id           = 65335
00000000 55 33 56 43 00 00 01 08 40 00 37 ff 49 6d 61 67  U3VC....@.7.Imag
00000010 69 6e 67 20 53 6f 6c 75 74 69 6f 6e 73 20 47 72  ing Solutions Gr
00000020 6f 75 70 00 00 00 00 00 00 00 00 00 00 00 00 00  oup.............
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00              ............    
packet_type  = cmd
command      = read-memory-cmd
size         = 12
id           = 65336
address      = 0x0000000000000044
size         =         64 (0x00000040)
00000000 55 33 56 43 00 40 00 08 0c 00 38 ff 44 00 00 00  U3VC.@....8.D...
00000010 00 00 00 00 cd 55 40 00                          .....U@.        
packet_type  = ack
command      = read-memory-ack
size         = 64
id           = 65336
00000000 55 33 56 43 00 00 01 08 40 00 38 ff 43 4d 56 34  U3VC....@.8.CMV4
00000010 30 30 30 20 4d 6f 6e 6f 00 00 00 00 00 00 00 00  000 Mono........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00              ............    

(process:5214): Aravis-CRITICAL **: 13:45:32.377: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed

(process:5214): Aravis-CRITICAL **: 13:45:32.377: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
[ArvDevice::get_integer_feature_value] Node 'SensorWidth' is not an integer
[ArvDevice::get_integer_feature_value] Node 'SensorHeight' is not an integer
Genicam: FAILED Invalid Genicam XML data
Sensor size read: 0 0

ben2789, 2018-10-23:
No image is displayed when using the arv-viewer tool either. Although it is correctly detecting the resolution , and some of the setable parameters are present in the drop down menu (exposure, and gain).

EmmanuelP, 2018-10-23:
arv-test is an incomplete test tool that is still useless. I should disable its compilation.

Could you attach the output of:

`arv-tool-0.6 features`

and the output of:

`tests/arv-camera-test -d all`

ben2789, 2018-10-23:
arv-camera-test

```./arv-camera-test
Looking for the first available camera
vendor name           = Imaging Solutions Group
model name            = CMV4000 Mono
device id             = (null)
image width           = 2048
image height          = 2048
horizontal binning    = 1
vertical binning      = 1
payload               = 4194304 bytes
exposure              = 15000 µs
gain                  = 1 dB
Frame rate = 89 Hz
Frame rate = 66 Hz
Frame rate = 67 Hz
Frame rate = 66 Hz
Frame rate = 67 Hz
Frame rate = 66 Hz
Frame rate = 67 Hz
Frame rate = 67 Hz
Frame rate = 66 Hz
Frame rate = 67 Hz
^CFrame rate = 66 Hz
Completed buffers = 754
Failures          = 0
Underruns         = 0
```
arv-tool-0.6 features

```
Imaging Solutions Group-00000000 (USB3)
Category: 'Root'
    Category: 'DeviceControl'
        StringReg: 'DeviceVendorName'
        StringReg: 'DeviceFamilyName'
        StringReg: 'DeviceModelName'
        StringReg: 'DeviceSerialNumber'
        StringReg: 'DeviceVersion'
        StringReg: 'DeviceFirmwareVersion'
        StringReg: 'DeviceManufacturerInfo'
        Integer: 'DeviceSFNCVersionMajor'
        Integer: 'DeviceSFNCVersionMinor'
        Integer: 'DeviceSFNCVersionSubMinor'
        Integer: 'DeviceManifsetXMLMajorVersion'
        Integer: 'DeviceManifsetXMLMinorVersion'
        Integer: 'DeviceManifsetXMLSubMinorVersion'
        Integer: 'DeviceGenCPVersionMajor'
        Integer: 'DeviceGenCPVersionMinor'
        Integer: 'DeviceU3VVersionMajor'
        Integer: 'DeviceU3VVersionMinor'
        Enumeration: 'DeviceTemperatureSelector'
            EnumEntry: 'Sensor'
        Float: 'DeviceTemperature'
        Float: 'DeviceTemperatureFahrenheit'
        Enumeration: 'DeviceLogLevel'
            EnumEntry: 'Verbose'
            EnumEntry: 'Debug'
            EnumEntry: 'Info'
            EnumEntry: 'Warning'
            EnumEntry: 'Error'
            EnumEntry: 'Silent'
        Command: 'DeviceReset'
        Boolean: 'ColumnCorrectionEnable'
        Boolean: 'RowCorrectionEnable' (Not available)
        Boolean: 'PixelDefectCorrectionEnable'
    Category: 'ImageFormatControl'
        Integer: 'Width'
        Integer: 'Height'
[... 102 lines cut ...]
            EnumEntry: 'Default'
    Category: 'RoiSequence'
        Enumeration: 'TableSelect'
            EnumEntry: 'Capture'
            EnumEntry: 'Preview'
        Integer: 'TableEntry'
        Integer: 'TableIndex'
        Command: 'CompileROITable'
        Enumeration: 'ROIControlMode'
            EnumEntry: 'ROI_Table'
            EnumEntry: 'Normal'
        Integer: 'TablePreviewLength'
        Integer: 'TableCaptureLength'
    Category: 'MultiSlopeExposure'
    Category: 'ISGCustom'
        Enumeration: 'LEDStatusIndicator'
            EnumEntry: 'On'
            EnumEntry: 'Off'
        Enumeration: 'FrameMetaData'
            EnumEntry: 'On'
            EnumEntry: 'Off'
        Enumeration: 'FrameBufferState'
            EnumEntry: 'Overrun'
            EnumEntry: 'Normal'
`

EmmanuelP, 2018-10-23:
The output of arv-camera-test seems to indicate streaming is correctly started, and frames are received without error.

Does arv-viewer works with the embedded Fake camera ?

ben2789, 2018-10-23:
interestingly no the viewer doesn't work correctly with any of my cameras (including fake), even cameras which work through a driver based on aravis. However the aravis based driver seems to work with my other cameras, even when the viewer does not. It does not however work with the LW-AL-CMV4000M. 

my system is running 18.04 4.023. I will investigate why the driver isn't working with this particular camera. It may be a fault of my installation and driver, rather than aravis.

Thanks.

ben2789, 2018-10-23:
arv-test-0.6 also fails on my functional camera. I have edited the aravis based driver to work with all cameras now. I do believe this was not an aravis issue. Sorry for wasting your time, thank you for the help debugging this.

closed 2018-10-23 by ben2789