AravisProject/aravis #207: Cross compilation for ARM v7

madStEvil, 2018-10-30:
Hi there,

I tried to cross compile the aravis library, but it doesn't work.
I have not much expirience to cross compile such a big project, so i hope someone can help.

After i cloned aravis from github,
i choosed the cross compiler from my buildroot enviroment as follows:
`export CC="/media/sdb1/trunk/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/bin/arm-linux-gcc"`

Than i run the autogen.sh script:
`./autogen.sh --host=arm-linux-gnueabi`

and build aravis with:
`make`

On the first build process i got errors, but after i added some missing packages (libxml2 and icu) to buildroot, it worked.

than i tryed to copy all the compiled libs from host (aravis/src/.libs/libaravis-0.6.so*) to the ARM Controller (/usr/lib),
and the demo snippets from host machine (like aravis/src/.libs/arv-tool-0.6) to the ARM Controller (/root) too.

But when i start the demo from ARM i get this bus error:

```
# ./arv-tool-0.6 list
Alignment trap: not handling instruction e8d4237f at [<b6b7b54e>]
Unhandled fault: alignment exception (0x001) at 0xb6ba4ddc
Bus error
```

There is also a libtool generated script (arv-tool-0.6) inside src directory with following description:

```
# The arv-tool-0.6 program cannot be directly executed until all the libtool
# libraries that it depends on are installed.
```
But i can't run this script under the ARM Controller because there is no compiler.
Also the script-pathes link to host directorys...
If i start the script from host it generates lt-arv-tool-0.6 successfully, but this file give me a bus error on arm too.

So i think i miss some libraries or include-pathes?
Is there a quick howto to cross compile aravis?

Thank you very much, 
and sorry, my english is not very well!

CaffeMrDe, 2019-03-20:
I advise you can check the way of  the compile the aravise. Look the document or readme.

highperformancecoder, 2019-08-15:
I have successfuly compiled the latest top of tree on ARM7. Latest aravis uses meson, so you need to set up a meson cross-file, as per meson cross compilation notes. I attach my meson cross-file, which is for the TI Arago SDK.

```
[binaries]
c = '/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/x86_64-arago-linux/usr/bin/arm-linux-gnueabihf-gcc'
cpp = '/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/x86_64-arago-linux/usr/bin/arm-linux-gnueabihf-g++'
ar = '/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/x86_64-arago-linux/usr/bin/arm-linux-gnueabihf-ar'
strip = '/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/x86_64-arago-linux/usr/bin/arm-linux-gnueabihf-strip'
pkgconfig = '/home/rks/usr/bin/arm-pkg-config'
cmake='/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/x86_64-arago-linux/usr/bin/cmake'
#exe_wrapper = 'qemu' # A command used to run generated executables.

[properties]
sys_root='/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/armv7at2hf-neon-linux-gnueabi/'

[host_machine]
system = 'linux'
cpu_family = 'arm'
cpu = 'armv7hl'
endian = 'big'

[paths]
prefix='/opt/ti-processor-sdk-linux-am57xx-evm-06.00.00.07/linux-devkit/sysroots/armv7at2hf-neon-linux-gnueabi/usr'

```

EmmanuelP, 2019-08-18:
Thanks @highperformancecoder 

I have pushed your cross file to master in /tools/meson-cross

@madStEvil please refer to meson cross compilation documentation: https://mesonbuild.com/Cross-compilation.html#changing-cross-file-settings

closed 2019-08-18 by EmmanuelP

highperformancecoder, 2019-08-19:
> mesonbuild.com/Cross-compilation.html#changing-cross-file-settings

Sure - just so long as its understood as being an exemplar, and will
need to be tweaked for any individual's circmstances.

semrade, 2023-01-14:
@highperformancecoder 
Hello, I am trying to cross-compile on a Linux host for arm64 bit and I am using meson.
The arm toolchain installed is arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-linux-gnu
I shared the issue I am facing in this post, do you have any idea, please?
https://github.com/mesonbuild/meson/discussions/11292
Thank you in advance ;)

highperformancecoder, 2023-01-14:
> mesonbuild/meson#11292

I'm not sure I'm able to help much, actually.

My experience of building Aravis on arm was on the Sitara chip, which
is an ARM 32 architecture. TI provided a complete cross-compilation
environment, with many popular libraries already built. I cannot
remember now whether glibc was provided already, or had to be built
from source. I don't remember any specific difficulty with glibc,
though.

The problem is that my client abandoned the Sitara chipset in early
2020 when we realised there was no way it was going to achieve the
bandwidth we needed the its PCI bus. We now use a USB attached DAQ,
hosted by x86_64 chips, which proved more amenable and simpler to
program.

So not only is my experience on a completely different chipset (ARM32
- Sitara specifically), with a vendor supplied development environment
that took a lot of the pain out of it, I also haven't touched that
environment for 3 years, and in fact have deleted it off my system, as
it is very large.

semrade, 2023-01-15:
@highperformancecoder 
Hello, thank you for the answer, yes I have the impression that TI offers a complete package which is not the case with ARM cross compiler. Ok, I'll try, and if I can do it I leave the solution here.
have a nice day and good luck with the new board :)