AravisProject/aravis #211: arv-camera-test tries to allocate enormous amount of memory

lifesciencemethods, 2018-11-23:


lifesciencemethods, 2018-11-23:
Hi,

I'm trying to get my ContrasTech camera to work. But, as shown below, the arv-camera-test function fails with an out-of-memory error while it tries to allocate an enormous amount of memory.

```
# ./arv-camera-test 
Looking for the first available camera

(arv-camera-test:5555): GLib-ERROR **: 17:15:21.287: ../../../../glib/gmem.c:135: failed to allocate 94740676804867 bytes
Trace/breakpoint trap
```

EmmanuelP, 2018-11-24:
Please rerun arv-camera-test with debugging enabled:

`arv-camera-test -d all:3`

lifesciencemethods, 2018-11-25:
Hi,

Ok, the problem seems to be different (arv-viewer hangs; but I can't reproduce the original problem); let me give some more details. I've rebooted the computer, inserted the camera into the USB port.

```
# dmesg
...
[ 2673.191345] usb 2-4: new SuperSpeed USB device number 3 using xhci_hcd
[ 2673.212121] usb 2-4: New USB device found, idVendor=2e03, idProduct=0001
[ 2673.212123] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2673.212124] usb 2-4: Product: Mars5000S-75uc
[ 2673.212125] usb 2-4: Manufacturer: ContrasTech
[ 2673.212126] usb 2-4: SerialNumber: 4H074A8PAK2417E
```

Running lsbusb gives this:
[lsusb.txt](https://github.com/AravisProject/aravis/files/2613184/lsusb.txt)

Running arv-camera-test now actually works:

```
# ./arv-camera-test 
Looking for the first available camera
vendor name           = ContrasTech
model name            = Mars5000S-75uc
device id             = (null)
image width           = 2448
image height          = 2048
horizontal binning    = 0
vertical binning      = 0
payload               = 5013504 bytes
exposure              = 10000 µs
gain                  = 0 dB
Frame rate = 82 Hz
Frame rate = 66 Hz
Frame rate = 64 Hz
Frame rate = 66 Hz
Frame rate = 64 Hz
Frame rate = 66 Hz
Frame rate = 65 Hz
Frame rate = 65 Hz
Frame rate = 65 Hz
Frame rate = 64 Hz
^C^CFrame rate = 64 Hz
Completed buffers = 731
Failures          = 0
Underruns         = 0
^C
```

And I can do this multiple times in succession with similar results.

Now I'm starting arv-viewer:

```
# arv-viewer
```

It takes a long time (10+ seconds) for the application to start. This is what it looks like eventually:

[image]

Now I click on the ContrasTech device:

[image]

And when I click on the "play" button (video acquisition), then the UI looks slightly messed up, as if it isn't updating (see blue bars at the side), and the fonts look slightly garbled:

[image]

In fact, the UI is not updating, which is evident from this picture (note that the application is running in the foreground):

[image]

Running top shows that arv-viewer is using about 100% of CPU.

Ok, so after a while I clicked the "cross" to close the window. The arv-viewer is not running anymore.

The arv-camera-test now seems to be working still.

```
# ./arv-camera-test 
Looking for the first available camera
vendor name           = ContrasTech
model name            = Mars5000S-75uc
device id             = (null)
image width           = 2448
image height          = 2048
horizontal binning    = 0
vertical binning      = 0
payload               = 5013504 bytes
exposure              = 10000 µs
gain                  = 0 dB
Frame rate = 53 Hz
Frame rate = 68 Hz
Frame rate = 65 Hz
^CFrame rate = 66 Hz
Completed buffers = 252
Failures          = 0
Underruns         = 0
^C
```

dmikushin, 2019-05-10:
Same corrupted viewer screen behavior for Mars USB3 camera for me here, also Ubuntu 18.04.

EmmanuelP, 2019-05-10:
Does the video view work if you select the Fake camera ?

dmikushin, 2019-05-10:
No, same display corruption.

Debugging shows the `buffer` pointer in the streaming code below is 0x0:

```
// in ../../src/arvuvstream.c lines:
 192                                     case ARV_UVSP_PACKET_TYPE_DATA:                                                                      >193                                             if (buffer != NULL && buffer->priv->status == ARV_BUFFER_STATUS_FILLING) {
...
```

dmikushin, 2019-05-10:
For the fake camera the buffer is not NULL, and I can see the arv buffer smoothly translates to gst buffer. However, in `snapshot_cb` this is what I have in the `data` for the fake camera:

```
(gdb) p data
$10 = 0x7fffe166e010 "3456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abcdefghijklmnopqrstuvwxyz{|}~\177\200\201\202\203\204\205\206\207\210\211\
212\213\214\215\216\217\220\221\222\223\224\225\226\227\230\231\232\233\234\235\236\237\240\241\242\243\244\245\246\247\250\251\252\253\254\255\
256\257\260\261\262\263\264\265\266\267\270\271\272\273\274\275\276\277\300\301\302\303\304\305\306\307\310\311\312\313\314\315\316\317\320\321\
322\323\324\325\326\327\330\331\332\333\334\335\336\337\340\341\342\343\344\345\346\347\350\351\352\353\354\355\356\357\360\361\362\363\364\365\
366\367\370\371", <incomplete sequence \372>...
```

Size is set 512x512, format is Mono8. But I wonder if those "...abcdef..." kind of data content is there by intention or is corrupted?

dmikushin, 2019-05-10:
Finally, I've dumped the packet right from the `case ARV_UVSP_PACKET_TYPE_DATA:` into a BMP file using awesome https://github.com/skeeto/bmp , and noticed the camera frame was captured perfectly. So, looks like gtk/gstreamer issue.

EmmanuelP, 2019-09-04:
Video display of a GStreamer pipeline is not really in a good shape currently. The only reliable sink I have found is gtksink, but it is not efficient, and use a lot more CPU power than xvimagesink. But xvimagesink is not usable in a wayland desktop.

I'm closing this issue. I will still try to find a better and efficient way to display a video embedded in a Gtk widget.

Thanks for the report.

Cheers.

closed 2019-09-04 by EmmanuelP