AravisProject/aravis #217: Bug in Aravis Viewer 0.6

Gauthamraju31, 2018-12-28:
The camera model I am using is VCXG-24C from baumer with Aravis 0.6. When I run the Aravis Viewer the first time, everything works fine when I try it again the viewer hangs for a minute and it no longer is able to detect the camera and even the software provided by baumer no longer detects the camera.

Following is the outcome of arv-test-0.6 command
`Testing device 'VCXG-24C' from 'Baumer'

(process:9916): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed

(process:9916): Aravis-CRITICAL **: arv_xml_schema_validate: assertion 'schema->priv->valid_ctxt != NULL' failed
Genicam: FAILED Invalid Genicam XML data
Sensor size read: 1920 1200
`

Gauthamraju31, 2018-12-28:
More Info - 

./arv-camera-test

Looking for the first available camera
vendor name           = Baumer
model name            = VCXG-24C
device id             = 700001878307
image width           = 1920
image height          = 1200
horizontal binning    = 1
vertical binning      = 1
payload               = 2304000 bytes
exposure              = 4000 µs
gain                  = 10 dB
gv n_stream channels  = 0
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 576 bytes
Frame rate = 44 Hz
Frame rate = 39 Hz
Frame rate = 39 Hz
Frame rate = 38 Hz
Frame rate = 39 Hz
Frame rate = 39 Hz
Frame rate = 38 Hz
Frame rate = 39 Hz
Frame rate = 39 Hz
Frame rate = 38 Hz
Frame rate = 39 Hz
Frame rate = 39 Hz
Frame rate = 38 Hz
Frame rate = 39 Hz
Frame rate = 39 Hz
Frame rate = 38 Hz
^CFrame rate = 39 Hz
Completed buffers = 663
Failures          = 0
Underruns         = 0

./arv-heartbeat-test

[GvDevice::new] Interface address = 192.168.1.228
[GvDevice::new] Device address = 192.168.1.53
[GvDevice::load_genicam] xml url = 'Local:Baumer_VCXG-24C_rev8C1F978CDBA81A241A1E50A9FA6D5DAA762A2538.zip;10370000;E290' at 0x200
[GvDevice::load_genicam] Xml address = 0x10370000 - size = 0xe290 - Baumer_VCXG-24C_rev8C1F978CDBA81A241A1E50A9FA6D5DAA762A2538.zip
[GvDevice::load_genicam] Zipped xml data
[ArvDevice::set_string_feature_value] Node 'GevGVSPExtendedIDMode' is not a string
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::create_stream] Number of stream channels = 1
[GvStream::stream_new] Packet size = 576 byte(s)
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
[... 148 lines cut ...]
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
Setting Gain from (null) to 10
^C[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)

Completed buffers = 890
Failures          = 0
Underruns         = 0
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)
[GvDevice::read_memory] Unexpected answer (0x0000)

pet110, 2019-01-16:
Hi Gautham,

I just tried to reproduce your issue... using the aravis 0.6.1 the camera connects/disconnects fine as far as I can see. I also did run the arv-test-0.6 with similar errors to yours, however the arv-viewer and the arv-test work fine for me.

Could you provide me with the steps to reproduce the error? Could you also provide me with the "Device Firmware Version" of your camera so i can check that this is up-to-date? You can find this information in the camera features using the Baumer Camera Explorer.

Peter

EmmanuelP, 2019-01-16:
Hi,

Just for your information, arv-test-0.6 is an unfinished test utility, not useful yet. Build of this tool is disabled in aravis 0.6.1.

Cheers,

    Emmanuel.

pet110, 2019-01-16:
Hi Emmanuel,

that's good to know, thanks for the info...

Gauthamraju31, 2019-01-17:
@pet110 The info you asked for -

Open first device 
     Device DeviceID:        devicemodul00_06_be_01_8a_9d
     Device Model:           VCXG-24C
     Device SerialNumber:    700001878307
     Device Vendor:          Baumer
     Device TLType:          GEV
     Device AccessStatus:    RW
     Device UserID:          cam1

Opened device - RemoteNodeList Information 
     Device AccessStatus:    RW
     DeviceSerialNumber:     700001878307
     DeviceManufacturerInfo: F:0200787B/C:020078B9/BL3.7:02000002
     DeviceFirmwareVersion:  CID:020001/PID:11165943
     GevCCP:                 ControlAccess
     GevCurrentIPAddress:    192.168.1.53
     GevCurrentSubnetMask:   255.255.255.0

 My Camera setup - 
 Baumer Camera ----> 10 gigabit switch  <---- PC

[image]

You can see the issue in this gif, it disappears after appearing first time

[image]

pet110, 2019-01-18:
Hi,

OK... the Firmware should be fine. I'm not able to reproduce the issue you have on my Ubuntu 18. So i would guess it is not a camera releated problem. Have you tried a different camera already? Which Platform and which Distri are you using?

Peter

pet110, 2019-01-21:
Hi,

it looks like we have a very similar setup. I see 2 potential differences.

I have used the 0.6.1 package and compiled without any special configuration parameters from http://ftp.acc.umu.se/pub/GNOME/sources/aravis/0.6/ ... did you use the same?
I have no Switch but have the camera connected directly maybe that could introduce the issue you describe?

Peter

pet110, 2019-04-10:
Hi @Gauthamraju31,
sorry that i could not help you and you had to go through the extra work... I'm the product manager for the Software here at Baumer. We are currently working on a Python interface which I hope to release in the coming months. I'll let you know when that is ready in case you are interested.

Best Peter

Gauthamraju31, 2019-05-01:
Hi @pet110 , thank you for your support and please let me know when the python api gets ready.

romainreignier, 2019-06-06:
Just for information, I do not have this issue with Ubuntu 18.04, aravis 0.6.1 and camera Baumer VCXG-124C.

EmmanuelP, 2019-09-04:
Looks like it is not an Aravis issue.

I'm closing this report. Don't hesitate to reopen if you think there is somtehing we can do in Aravis.

Cheers.

closed 2019-09-04 by EmmanuelP